realthunder/FreeCAD_assembly3 #1125: Subshape Binder causes crash when used with edges from Part workbench.

BeresfordRomeo, 2022-02-19:
[image]

Hi there. I am not sure if this is much of an issue but if I try to create a Subshape Binder in Part Design with edges from surface extrusions created in the Part workbench then FreeCad crashes (closes without warning). Interestingly it does not crash when I use Blend Curves from the Curves workbench. 

[image]

I am not suggesting that it should be possible to use edges from Part but just wanted to let you know about the crashing issue.

Thanks again for all your good work. It is so much appreciated and if I can help in any way I would be happy to do so.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

realthunder, 2022-03-21:
I can't reproduce the problem. Are there any error output? Maybe you can record a screen cast.

BeresfordRomeo, 2022-05-06:
Hi @realthunder 

I am so sorry I have not had the opportunity to give you more information on this. I will try to record a screencast when I can work out how, apologies for the delay.

luzpaz, 2022-05-06:
@BeresfordRomeo  Can you also test with LinkDaily?