realthunder/FreeCAD_assembly3 #1126: [Issue] Pressing = does not open function anymore.

lmunier, 2022-02-19:
Hello,

Everything is in the title. I am working with the following version of FreeCAD :

OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

When I want to put a new quote, I have to press on the function button to have the expression editor, pressing = does not work anymore, as you can see on the following screenshot.
[image]

brandonbeaumia, 2022-02-19:
Seems to be working as intended for me... did you accidentally clear or change the setting? Edit > Preferences > Display > UI >  Editor Trigger Shortcut

OS: Pop!_OS 20.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

lmunier, 2022-02-19:
Hi @brandonbeaumia and thank you for your help.

I did not know about this parameter, it was the default value : = and by trying different ones I realized that it works but not everytime. Let me explain :

Keys that works (tested) : E, D, 1
Keys that does not work (tested) : =, Shift+=, Shift + D, Space, None

I do not know if it comes from my keyboard's layout, since I have a qwertz one and I need to press Shift + = to have a = sign.

brandonbeaumia, 2022-02-19:
I see; thank you for testing more.

I can also confirm the following:

Ctrl + alphanumeric: works fine.
Ctrl + symbol: works fine.
Alt + alphanumeric: works fine.
Alt + symbol: works fine
Shift + alphanumeric: does not work.
Shift + symbol: does not work.

So it seems to be a problem with the shift key. This sort of makes sense as we are focusing a text field while trying to perform the shortcut. I went back to the 2021-10-15 to check but found the same behavior. Did this work for you in a previous build?

lmunier, 2022-02-19:
You're welcome, I am happy to give a bit of help. Thank you to you to help finding why it does not work as expected.

It worked in an older version, this is why I thought about an issue. The version where it worked was the following (and it worked also on the stable one from the same date) :

OS: Ubuntu 20.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

I was able to test the same shortcuts as you and noticed almost the same behavior :

Ctrl + alphanumeric: works fine.
Ctrl + symbol: works fine. (if it is not a used shortcut like Ctrl + v)
Alt + alphanumeric: works fine.
Alt + symbol: works fine
Shift + alphanumeric: does not work. (= does work this time, on my keyboard it is Shift + 0)
Shift + symbol: does not work.

And also, on this version, even if the shortcut is overwritten, the default one also work. It means that the two shortcuts work to open the expression editor.

brandonbeaumia, 2022-02-19:
Interesting...I wonder if = was hardcoded in the previous build and setting the preference to something else only _added_ that shortcut. If the hardcoding was removed it would now be relying on the (apparently bad) shift key behavior. Bummer. Hopefully @realthunder has time to look into this, but in the short term your best bet is probably to just change the shortcut to 0 and omit the shift key for now.

bambukouk, 2022-02-20:
Just to confirm :+1: 
Happens to me intermitently as well
Only way to deal with it (so far) is to use keyboard return key (instead of clicking the mouse)
Haven't managed to replicate reliably... 
Can't include hardware/about info (away from PC)
using latest daily on Linux

closed 2022-05-18 by realthunder