realthunder/FreeCAD_assembly3 #1127: File save as adds original filename to internal references

GinTonic99, 2022-02-19:
My models normally are in one file only. From time to time I save a status under a new name. When doing so it happens for some internal references that the original file name is added in front of some internal references e.g. to a spreadheet cell. The file attached to #1067 can serve as example. A really full recomputation shows some references broken because they are pointing to a previous filename.

currently in use the followong version, the error was observed already before in previous versions.

OS: Debian GNU/Linux bookworm/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Austria (de_AT)

realthunder, 2022-03-19:
Can you please elaborate your steps to reproduce the error? I didn't find anything wrong after saving the file to a new name.

GinTonic99, 2022-03-20:
If I remember correctly, I saved the file under a different name in a new directory and made some changes in hte new file. Then after an automatic recalcuation an error appeared caused by the reference to the file with the "old" name. I am not anymore sure if these references are extended during the save as operation or later during some chenges and recalculation. Next time the effect appears I will save the intermediate status.

There is a similar effect when I copy e.g. a body in part design with ctrl-c ctrl-v. This operation addds the filenames to the references of the copy. If such a file is then saved under a new name in a different directory, the references break. 

I enclose a file showing this copy effect. The original file name was _Magnethalter_04_ and was saved under _test_. The body Aussen-001 is a copy of Aussen now having the references with the old filename.

thank you for attending the issue

[image]

[test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8311783/test.zip)

closed 2022-03-21 by realthunder