realthunder/FreeCAD_assembly3 #1128: pie menu don't work on mac

Glycocalix, 2022-02-19:
Hello, 
I'm using Freecad with a great pleasure. Recently I tried to use Pie menu on my Mac (intel chip's) and OS Big Sur (11.5.1).
I've created a global pie menu inserting sketcher tools. I try to run de pie menu by using short cut F18, F19 or F12. when I draw a sketch I press the shortcut but the pie menu doesn't appears
[image]
r

free cad version is 2022.111 (30764 Git)

regards

luzpaz, 2022-02-19:
Hi, please always add full [About](https://wiki.freecad.org/About) info. Thanks

Glycocalix, 2022-05-10:
OS: macOS 11.5
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)