realthunder/FreeCAD_assembly3 #1129: Point error

hamza00789, 2022-02-20:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)
  
Not considering the point as an element. The problem can be seen in the video.   
  

https://user-images.githubusercontent.com/92737149/154846620-14a1f1e2-cd0c-491b-92f6-4556e43dbd6d.mp4

Rabbit-sk, 2022-02-22:
Maybe related: #1049 #1076

realthunder, 2022-03-19:
Please post a sample file showing the problem.

hamza00789, 2022-03-26:
> Please post a sample file showing the problem.   

The section command is used on two sketches Because the result of the section is a point, it is not coming. 

[image] 

[image]

[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8354996/1.zip)

realthunder, 2022-03-26:
How is this related to the problem in your opening post? Section only works for solid.

hamza00789, 2022-03-26:
> 

This is not the case with Section 2d Also works for geometry.

hamza00789, 2022-03-26:
> How is this related to the problem in your opening post? Section only works for solid.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US) 

https://user-images.githubusercontent.com/92737149/160239259-a2615d75-c967-4eb1-8dac-00c5fd87bcbc.mp4

hamza00789, 2022-03-26:
Video posted on this page does not appear to be playing.

realthunder, 2022-03-26:
My mistake. It does work with older OCC. With 7.6 it's broken for some reason.

realthunder, 2022-03-26:
But still, I don't see how it's related to your OP.

hamza00789, 2022-03-26:
Because the result is a point, I thought maybe it had to do with it.

hamza00789, 2022-03-26:
Maybe these are two different issues.

hamza00789, 2022-03-26:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.118.30769 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3b620b713e24c9637db8ea6ad1f1ee88ef13367a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

I think the point is the problem and only the point fails to come. Point Command is not working even in the Draft workbench.  

https://user-images.githubusercontent.com/92737149/160240661-c1f669ad-7c38-448b-913f-c87f6db1946a.mp4

realthunder, 2022-03-26:
I can't see the last two videos. I can see the first video in your OP. Can you please recreate a file showing the problem in OP and post it here?

hamza00789, 2022-03-26:
> I can't see the last two videos. I can see the first video in your OP. Can you please recreate a file showing the problem in OP and post it here?

The problem you can see in the first video above. I just noticed that when the file is closed and the file is reopened, the point appears but is not in the right place. 
 
[image] 
[image] 

[A 1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8355614/A.1.zip)

hamza00789, 2022-03-26:
I think the point is the problem and only the point fails to come. Point Command is not working even in the Draft workbench.   

[5.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8355673/5.zip)

closed 2022-03-28 by realthunder