realthunder/FreeCAD_assembly3 #1130: Loss of workbenches list

jpg87Fr, 2022-02-20:
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.213.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.02.13-edge)
Hash: ce7fe6dc1c51f6074b58d125342fc3615b0a564f
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.5.2
Locale: French/France (fr_FR)

Hello I am currently testing the snap version of February 13 and I have a small problem :
At startup, I have my list
[image]
if I want to customize my list, I get this
[image]
when I close the window, no more workbench
[image]
I have to re-enter them manually in the settings editor (or in link.user.cfg).
[image]

luzpaz, 2022-03-13:
@jpg87Fr still an issue?

jpg87Fr, 2022-03-14:
Hello and thank you very much.
Works fine in the March 07 Snap release.

closed 2022-03-14 by jpg87Fr