realthunder/FreeCAD_assembly3 #1131: Wkiki build instructions out of date?

J-Dunn, 2022-02-21:
Are these build instructions on the wiki page still valid. They seem to be over a year since modification and still referring to slvs in the tree. 

https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction

It would be helpful to have current build instructions.

luzpaz, 2022-03-13:
@realthunder where is the most up to date info, if you want I can update this for you

J-Dunn, 2022-03-13:
I don't think you understood the subject of this issue.  I was not seeking info on lasted PRs and bugfixes. The point was that the current build instructions, as linked, do NOT seem to relate to the current state of the code.

luzpaz, 2022-03-13:
*Perhaps realthunder had another location he was maintaining the build instructions at and forgot to update this one* :arrow_left:  That was the assumption I was under

realthunder, 2022-03-15:
It is indeed outdated. There is now pip package for the solver backend. `pip install py-slvs`.

@luzpaz Who's in charge of the Add-on repository? Assembly3 will prompt for solver installation, so it can now be installed through the add-on manager.

luzpaz, 2022-03-15:
@realthunder I have permissions to manage the Addon repo. 
Shall we open a new ticket to discuss this?
Questions to address: 
* can A3 run on master? If not then we need to indicate that via the Addon Manager. 
* [`package.xml`](https://wiki.freecadweb.org/Package_Metadata) is needed (with a dependencies section)

J-Dunn, 2022-03-15:
That would be nice if Addon Manager could deal with this but there are about 4000 diffs between thunders branch and master, AFAIK it's a long way from working fully on master. 

Wouldn't the first thing to address this ticket be to correct the build instructions so they at least give correct instructions for the current code base?

luzpaz, 2022-03-18:
@realthunder please advise https://github.com/realthunder/FreeCAD_assembly3/issues/1131#issuecomment-1068072969

realthunder, 2022-03-19:
> * can A3 run on master? If not then we need to indicate that via the Addon Manager.

Yes, it can run on master.

> * [`package.xml`](https://wiki.freecadweb.org/Package_Metadata) is needed (with a dependencies section)

I'll add this file soon.

J-Dunn, 2022-03-19:
Silly me , I was reading the information provided by **this project** to say that was not possible. 

> At the moment of this writing, Assembly3 only works with a forked FreeCAD [branch](https://github.com/realthunder/FreeCAD/tree/LinkStage3). You need to first checkout this branch and [build](https://github.com/realthunder/FreeCAD/tree/LinkStage3#compiling) it yourself.

Is there any reason not to address the subject of this post and correct that now incorrect statement that I need the thunder fork to run A3 and to correct the build instructions?  I downloaded and build the fork next to my master branch and was running that for using A3. It seems that was a total waste of time since I could have just used master. 

There's some really impressive work here, a few  minutes correcting the basic information about the project would seem to be well worth it.

realthunder, 2022-03-19:
I have actually clarified these facts in the [read me](https://github.com/realthunder/FreeCAD_assembly3#installation) page long time ago. Anyway, I have just edited the build instruction to avoid future confusion.

J-Dunn, 2022-03-19:
I started off with the readme, went to _Installation_ and found this: 

> If you want to build everything yourself, please check out the build instruction [here](https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction).

At that point I clicked on "here" which was what I was looking for, since I was not interested in the distro specific instructions and windows. 

Thanks for updating the build instructions. Since you say master is not always up to date with latest A3 build I guess I did not waste my time cloning A3 master after all :)

 It's been a few years since I tried A3, it'll be great to see how it's improved.

closed 2022-03-19 by J-Dunn

luzpaz, 2022-03-19:
I'll add A3 to the Addon Manager today