realthunder/FreeCAD_assembly3 #1134: Pad  type "to first" and "to last" is not working properly

mwlandolsi, 2022-02-23:
Hello,
unlike last version of FreeCAD daily (15-10-2021), the pad type "to first" is not working properly.
**This is what we got in the previous version 15-10-2021 **
[image]

In Pad parameters, when selecting Type : "to first" or "to last"  with both cases "check face limit" checked or not, there is not the expected result.

When selecting "to first", the first surfaces are omitted,
[image]

When choosing "to last" the last face is not reached.

[image]

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

realthunder, 2022-02-24:
Can you please attach the file your trying?

hamza00789, 2022-02-24:
> Can you please attach the file your trying?  

I have a similar problem. 
 
[image] 

[image]
 
[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8131533/1.zip)

mwlandolsi, 2022-02-24:
Here is the file with tow bodies one with pad to first  and the other with pad to last

[pad to first last.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8132578/pad.to.first.last.FCStd.zip)

realthunder, 2022-03-18:
The above commit should fix the problem. It seems that you use `up to face` extrusion a lot. Can you please create some test suite for this feature.

mwlandolsi, 2022-04-21:
Thanks for your fixes. Sorry to answer in late because I was very busy these days.
Padding up to face, up to first, up to last, draft and thickness (2d and 3D) are commonly used in plastic parts design. So I use them, but with a lot of cares in this field.
Let's start for example by padding some ribs.
[image]

The attached file contains two types of  bodies that contain ribs created with "up to first" and "up to last" pad function.

Most of the common cases looks fine except with elliptic and B-Spline. There is also a small issue with the preview of "pad up to last" and check face limit is deactivated : the preview of the pad is created with the first surface like the condition pad up to first.

[image]

 So here, what is meant by up to last ? The last surface or the last material wall? If the last surface, then here the preview does not correspond to the definition of "up to last" in other ways, it  corresponds. 
With elliptic shapes  with thickness, it was not possible to create a pad in no case.
This is all what I have now. I will report some other cases if I find.
[test pad up to.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8527479/test.pad.up.to.FCStd.zip)