realthunder/FreeCAD_assembly3 #1136: What's the green eye in treeview ?

girow, 2022-02-24:
OS: ManjaroLinux
Word size of FreeCAD: 64-bit
Version: FreeCAD-asm3-Stable-Conda-Py3-Qt5-20211015-glibc2.12-x86_64.AppImage
Build type: Stable
Branch: LinkDaily
Locale: Japanse/japan (ja_JP)

[image]
Thank you for your develop.I fully enjoy linkstage Ver.

What is the green eye icon in treeview ? The green eye icon item can't be controlled visibility.

When FreeCAD restarts, the part and the sketcher remain visible.

Is there a good solution?

realthunder, 2022-02-24:
Yes, it is a known bug, and has already been fixed. The green eye marks the `Show on top` object. You can work around it for now by pressing `Ctrl + Shift + Space` when there is no selection. This clears all `Show on top` objects.

girow, 2022-02-24:
I'm sorry, I didn't see the bug information. 

And thank you for suggesting a solution ! I will try it !

luzpaz, 2022-02-24:
Closing ticket

closed 2022-02-24 by luzpaz