realthunder/FreeCAD_assembly3 #1138: The 'Fusion' is not working properly.

hamza00789, 2022-02-25:
When I use Fusion to make two solids one solid, the result is bad.  
I think this problem was present in all the older versions of freecad.

[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8141443/1.zip)

hamza00789, 2022-02-25:
[2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8141646/2.zip)

hamza00789, 2022-02-27:
Causes of problems 

These problems arise when the line is coincident with the point and the line is tangent with the circle. If a small distance is created between them, for example, a distance of 0.1mm between line and point. In this case, the fusion works perfectly. I think you already know about this problem. 
 
This situation creates a problem. 
 
[image] 
 
result   
 
[image] 

[image] 
   
result  
 
[image] 

In this case, the problem does not arise.  
 
[image] 
 
result 

[image] 

[image] 

result 

[image]
 
files 
[2 files.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8148591/2.files.zip)

hamza00789, 2022-02-28:
I think you already knew about this problem, will there be a solution in the future? Can you tell me something about that?

realthunder, 2022-03-18:
It's caused by seam edge touching other edge. Seam edge is defined as two edges of the same face overlaps. OCC is really bad at handling seam edges. You can work around this by rotating the shape a bit to avoid seam edge touching.

hamza00789, 2022-03-18:
Can OCC-related problems be solved? Is there any hope for a solution in the future?