realthunder/FreeCAD_assembly3 #1139: The children automatically leave the group.

hamza00789, 2022-02-27:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

Sometimes when I move things inside the group, the kids of some things get out of the group.  
For example, I inserted the blend surface into the group and the offset child extrude automatically leave the group. Sometimes a lot of children get out of the group that way.              
I see this as a problem that needs to be addressed.    

[MP4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8148410/MP4.zip) 
File 
[E 1 00.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8148412/E.1.00.zip)

realthunder, 2022-03-02:
Can you remember how you created the Part container and the Group? The Part container is special. The `Part.Group` property is supposed to contain all its (grand)children automatically, but the tree will only show those child objects that is not claimed by other children.

For example, if you create have a fusion of a cube and a cylinder, and you add that fusion to a Part container, it will automatically add the cube and cylinder as well. So the `Part.Group` property will show all three objects. But the tree view only shows the fusion object under the Part container. This is the expected behavior.

However, in your attached file, the `Part.Group` only contains the `Group` and `Blend_Surface004`, not any of the children in `Group`. This is not supposed to happen. When add `Blend_Surface004` to `Group`, the `Extrusion` object is automatically added to the Part container, because `Blend_Surface004` depends on `Extrusion`. Not only `Extrusion`, as you check `Part.Group`, it contains all the dependency of `Blend_Surface004`. And this is the expected behavior.

luzpaz, 2022-04-11:
@realthunder example file and clip were added to OP

realthunder, 2022-04-12:
I have checked your file and provide the explanation. My question remains. I want to know `how you created the Part container and the Group`. Did you type commends in the Python console to change the `Group` property of the container? I am asking because the part container you have in this file (and many other files of yours) is invalid. If you add the components to the container by normal drag and drop, you are not supposed to get this type of error.

hamza00789, 2022-04-12:
This is how I start each of my files. First I use a part container. Then I activate the part container. Then I create a group. Then I make whatever I want And then I keep throwing things in group. You can see in the given video I made a sketch and threw it inside the group then extruded it and threw it inside the group. 

 
[mp4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8471493/mp4.zip) 

That's why I asked if the group could be kept constantly moving. Within this issue  #1121

hamza00789, 2022-04-12:
Did you want to ask this?

hamza00789, 2022-04-12:
Doesn't know about the Python console and has never used it.

hamza00789, 2022-04-12:
Another problem I often encounter. It may have something to do with this problem. 

When I select from the tree, the selection is not indicated in 3D. But when I make a choice in 3D, the tree indicates a choice.  
 
[MP4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8472539/MP4.zip)

luzpaz, 2022-04-12:
> Doesn't know about the Python console and has never used it.

https://wiki.freecad.org/Python_console

realthunder, 2022-04-14:
> Another problem I often encounter. It may have something to do with this problem.

Can you please create a screencast duplicate the steps you did to create this file, the file with the non-responsive tree selection?

hamza00789, 2022-04-14:
> Can you please create a screencast duplicate the steps you did to create this file, the file with the non-responsive tree selection? 

I did not understand you.
This problem occurs when a father is thrown into the tree. In this case, it becomes impossible to choose anything from the tree. But if the child is picked up and thrown into a tree and placed back in the group, then the selection process begins. You can see the problem in the video. 

https://user-images.githubusercontent.com/92737149/163302964-76a2d042-25b1-4628-be9e-05843fcd6dc4.mp4  

[file.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8485832/file.zip)

realthunder, 2022-04-14:
Now I see why many of your files contain invalid Part containers. It is because you are dragging out objects that are not supposed to be dragged out. Those warnings are for that exactly. Those messages used to be printed out as errors, the the container would auto correct itself by removing the depending `Extrude` object out of the container the moment you drag the sketch out. Although this enforces a valid Part container, it is proved to cause more harm in many other use cases. I probably should revert to this behavior for drag and drop action only.

The correct way to do this for now, is to drag out both sketch and extrude, and then add back extrude only. Please note that the workflow you are doing is very close to what `PartDesign` is offering, but you try to perform it using `Part` workbench which is not designed for this, hence the quirkiness here and there.

hamza00789, 2022-04-14:
> Now I see why many of your files contain invalid Part containers. 

Can you fix the selection process?           

Because the second problem is that parents get out of the group. This problem can be solved in one way if things are inserted into the group through group property.      

https://user-images.githubusercontent.com/92737149/163344036-75824ddc-482e-4921-8123-849488a9f581.mp4             

I find it much better to look at the details of the previous work and work ahead. That's why I put things in groups. but when I don't need to see a dad I pick him up and put him in a tree. It also reduces the length of the tree. And the benefit of that is that I get to see the details of the workflow. I have used software in the past Because of this, I have a habit of working this way and I think this is the best way (workflow).

hamza00789, 2022-04-17:
Do you think it is appropriate to solve this problem? Can you solve it? Will you solve it?

closed 2022-04-20 by realthunder

realthunder, 2022-04-20:
Add two options to allow fine tune the normal group, and App::Part group behavior.

You can disable `Claim all children` to auto hide child object in a normal group that is also claimed by other child object. You can also enable `Allow cross coordinate links` to not auto remove object in some case, but use this option with caution. I strongly suggest you try the PartDesign workbench. Part features are not really designed for this type of workflow.

[image]