realthunder/FreeCAD_assembly3 #1140: question - "hint" position relative to cursor position

bambukouk, 2022-02-28:
Is it possible to adjust position of the hint window relative to the cursor?
As it is set currently it keeps getting in a way of mouse cursor action.
Please.

realthunder, 2022-02-28:
Can you please make a screen cast showing the problem? And I don't quite understand your notion of 'relative to the cursor'. Maybe a picture showing your idea?

bambukouk, 2022-02-28:
Sorry, should have thought about doing it before posting ...
[image]
As you can see I have moved over `animated_main_axle_front` on a way to the next body below, but I can neither see it or pick it because the hint window is in the way. If it could be moved slightly out of the way of the mouse cursor (say to the right and up?) than it would help. Please?

bambukouk, 2022-02-28:
and, in this example it works well:
I can see and select both items in the tree because hint window is slightly out of the way of my mouse cursor.
I guess it is something to do with the time hint window is visible as well?
I will try to shorten it to see if it helps?
[image]

bambukouk, 2022-03-02:
```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)

```
which is based on Ubuntu 20.04 LTS, so hopefully whatever you did? will work for me as well, thank you :+1: