realthunder/FreeCAD_assembly3 #1143: Tool library settings have vanished - how to recover/rebuild?

mnswamp1, 2022-03-02:
I've been working from the same file for the past few weeks, and today I tried to add a new body, sketch, pad and path job.  When I tried to add a tool to the job, it apparently crashed and now the tool library is blank - when I try to add a tool to the job, I get:
09:07:57  Traceback (most recent call last):
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathJobGui.py", line 874, in toolControllerAdd
09:07:57      tools = PathToolBitGui.LoadTools()
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBitGui.py", line 269, in LoadTools
09:07:57      return [PathToolBit.Factory.CreateFrom(foo) for foo in GetToolFiles(parent)]
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBitGui.py", line 269, in <listcomp>
09:07:57      return [PathToolBit.Factory.CreateFrom(foo) for foo in GetToolFiles(parent)]
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBit.py", line 416, in CreateFrom
09:07:57      bit = Factory.CreateFromAttrs(data, name, path)
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBit.py", line 403, in CreateFromAttrs
09:07:57      obj = Factory.Create(name, attrs['shape'], path)
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBitGui.py", line 179, in Create
09:07:57      tool = PathToolBit.ToolBitFactory.Create(self, name, shapeFile, path)
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBit.py", line 425, in Create
09:07:57      obj.Proxy = ToolBit(obj, shapeFile, path)
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBit.py", line 150, in __init__
09:07:57      self._setupBitShape(obj)
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBit.py", line 300, in _setupBitShape
09:07:57      (doc, docOpened) = self._loadBitBody(obj, path)
09:07:57    File "C:\FreeCAD_RT_Daily\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118\Mod\Path\PathScripts\PathToolBit.py", line 247, in _loadBitBody
09:07:57      doc = FreeCAD.openDocument(p, True)
09:07:57  TypeError: argument 1 must be str, bytes or bytearray, not None

I had created 10 other jobs in this file prior (and cut plywood) for this file.  

Thanks

realthunder, 2022-03-18:
Is it possible to share your file?

mnswamp1, 2022-03-18:
Not any more.  I wound up uninstalling wiping directories and config files
and reinstalling.  I am now wondering if there was a cross-setting between
the 'stock' v.19 and Assembly3 versions.  OK to close record at this point.

I have a couple of repeatable findings in Path to write up, but I'm good on
this one.

Thanks

Marsh Jones
"Geek by profession, coach by avocation, maker by desire, sports-hack by
obsession"
*President, **Twin Cities Paddlesports <https://tcpaddlesports.com/>*
*St Paul, MN*
*B <https://mnswamp.blogspot.com/>log
<https://mnswamp.blogspot.com/> Trailer Build
<https://swampwerks-teardrop.blogspot.com/>*

closed 2022-04-13 by luzpaz