realthunder/FreeCAD_assembly3 #1147: no solver backend found

spjt, 2022-03-11:
Using ChromeOS with Crostini:

On latest AppImage (FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage), I receive an error when starting: 

1.80357 <asm3.main> init_gui.py(14): no solver backend found

I do not receive this error when using the previous version (FreeCAD-asm3-Daily-Conda-Py3-Qt5-20211015-glibc2.12-x86_64.AppImage)

bambukouk, 2022-03-13:
```
NOTE! For better compliance with licensing restrictions, the SolveSpalce Python binding py-slvs is no longer bundled with the release images.
When launching Assembly3 workbench, or opening some file containing Assembly3 assemblies, you will be prompt to install py-slvs. The package is to be installed via pip as a user package, so it only need to be done once. If you it reports install failed, try restart FreeCAD

IMPORTANT! If you have manually installed Assembly3 workbench in your FreeCAD user directory, you may need to manually sync the repository to the latest in order to get the installation prompt.

```
from:
https://github.com/realthunder/FreeCAD_assembly3/releases

closed 2022-03-19 by luzpaz

Thovthe, 2022-04-11:
What is the licensing issue that this is supposed to solve?  Both freeCAD and solvespace are GPLv3

https://github.com/solvespace/solvespace/blob/master/COPYING.txt
https://github.com/realthunder/FreeCAD_assembly3/blob/master/LICENSE

luzpaz, 2022-04-11:
Incorrect, FreeCAD is LGPLv2.1

Thovthe, 2022-04-11:
So it is.

Thovthe, 2022-04-12:
These licenses are compatible though.
See:
https://www.gnu.org/licenses/license-list.html#GPLCompatibleLicenses
https://www.gnu.org/licenses/gpl-faq.en.html#AllCompatibility
And maybe this:
https://www.gnu.org/licenses/license-compatibility.html

luzpaz, 2022-04-12:
really don't know the details but that the addon manager is the way we can use GPL code within FreeCAD. Because it's not bundled in to the FreeCAD installation and instead pulled in post-install. If you have more questions, feel free to direct them to the FreeCAD forum.

Thovthe, 2022-04-14:
But it is the workbench pulling it in and Solvespace and the workbench are both GPLv3.  So why would it need to be unbundled from the workbench?

Thovthe, 2022-04-15:
>It does bring up a secondary issue, since A3 is GPLv3 can it be bundled with Link branch ?

Probably, but if we go with the assumption that the binding can't be bundled because it's GPLv3 then the workbench can't without being relicensed, and if other people have contributed that may be easier said than done.