realthunder/FreeCAD_assembly3 #1150: Linux Version crashes when opening file created with Windows Version

hansfbaier, 2022-03-12:
I created the attached file with Windows and when I try to open it with the Linux AppImage,
I get the following crash:
```
$ /opt/FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage PcieRiserCase.FCStd1 
/tmp/.mount_FreeCAUpTscs
FreeCAD 0.20, Libs: 2022.111R30764 (Git)
././././././././././././/../share/libdrm/amdgpu.ids: No such file or directory
A3 available
2.034622 <asm3.main> init_gui.py(14): no solver backend found
File format not supported: PcieRiserCase.FCStd1 
<Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of PcieRiserCase#Compound
<Part> ViewProviderExt.cpp(2149): Cannot compute Inventor representation for the shape of PcieRiserCase#Compound001
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x430c0) [0x7f99577470c0]
#1  0x7f993c70a2d5 in llvm::ConstantFoldCastInstruction(unsigned int, llvm::Constant*, llvm::Type*) from /opt/amdgpu/lib/x86_64-linux-gnu/libLLVM-12.so+0x225
#2  /opt/amdgpu/lib/x86_64-linux-gnu/libLLVM-12.so(+0x944d0a) [0x7f993c732d0a]
#3  /opt/amdgpu/lib/x86_64-linux-gnu/libLLVM-12.so(LLVMBuildBitCast+0x74) [0x7f993c749694]
#4  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xcff7bd) [0x7f994729d7bd]
#5  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xd065ab) [0x7f99472a45ab]
#6  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xd06a22) [0x7f99472a4a22]
#7  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xcb3c80) [0x7f9947251c80]
#8  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xc5e969) [0x7f99471fc969]
#9  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xc61137) [0x7f99471ff137]
#10  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xc6702e) [0x7f994720502e]
#11  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xbd3656) [0x7f9947171656]
#12  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xbcc042) [0x7f994716a042]
#13  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xbcc9fb) [0x7f994716a9fb]
#14  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0xbccf4b) [0x7f994716af4b]
#15  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x32ef86) [0x7f99468ccf86]
#16  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x32b889) [0x7f99468c9889]
#17  /opt/amdgpu/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x4ed131) [0x7f9946a8b131]
#18  0x7f9957451f43 in SoMarkerSet::GLRender(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x5a3
#19  0x7f9957309a5b in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x3db
#20  0x7f9957309a5b in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x3db
#21  0x7f99572d8eb7 in SoAnnotation::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x97
#22  0x7f99570e90f7 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x67
#23  0x7f99572a0992 in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x82
#24  0x7f995730da81 in SoSwitch::doAction(SoAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x2b1
#25  0x7f9959dbf246 in SoFCSwitch::doAction(SoAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x4d6
#26  /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so(+0x8d02ae) [0x7f9959df22ae]
#27  0x7f9959df2d95 in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0xa75
#28  0x7f9959df1fca in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x8a
#29  0x7f9957308d6d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x3ed
#30  0x7f9959ddd151 in Gui::SoFCUnifiedSelection::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x61
#31  0x7f9957308d6d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x3ed
#32  0x7f995742f5c6 in SoShadowGroupP::GLRender(SoGLRenderAction*, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x256
#33  0x7f9957308d6d in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x3ed
#34  0x7f99570e90f7 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x67
#35  0x7f9959dff55c in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x8c
#36  0x7f99570e9a82 in SoAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x5c2
#37  0x7f9959dfcea9 in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x159
#38  0x7f9959dfcf19 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x29
#39  0x7f99570f0291 in SoGLRenderActionP::renderSingle(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x271
#40  0x7f99570f077e in SoGLRenderActionP::render(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x11e
#41  0x7f9959dff55c in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x8c
#42  0x7f99570e9dea in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x2ea
#43  0x7f9959dfd28a in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x2a
#44  0x7f99573d96c0 in SoRenderManager::actuallyRender(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x70
#45  0x7f99573d9877 in SoRenderManager::renderSingle(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0xc7
#46  0x7f99573d9eab in SoRenderManager::render(SoGLRenderAction*, int, int, int) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/../lib/libCoin.so.80+0x9b
#47  0x7f9959ede317 in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x387
#48  0x7f9959e8d80d in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x31d
#49  0x7f9959e93a95 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x45
#50  0x7f995873b95d in QWidget::event(QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
#51  0x7f99587a4db7 in QFrame::event(QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Widgets.so.5+0x27
#52  0x7f995891febe in QGraphicsView::viewportEvent(QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Widgets.so.5+0x456
#53  0x7f9957ca8019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Core.so.5+0xcd
#54  0x7f99587107e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#55  0x7f995871631d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#56  0x7f9959a60999 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x89
[... 39 lines cut ...]
#96  0x7f9959a60999 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x89
#97  0x7f9957caa606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Core.so.5+0x86
#98  0x7f9957caa83e in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Core.so.5+0x1da
#99  /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Widgets.so.5(+0x3657a2) [0x7f99589017a2]
[... cut ...]
#114  0x7f9957caae8f in QCoreApplication::exec() from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib/libQt5Core.so.5+0xfd
#115  0x7f99599d1c03 in Gui::Application::runApplication() from /tmp/.mount_FreeCAUpTscs/usr/bin/../lib64/libFreeCADGui.so+0x443
#116  /tmp/.mount_FreeCAUpTscs/usr/bin/FreeCADLink(+0x473d) [0x55e2c861b73d]
#117  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f99577280b3]
#118  /tmp/.mount_FreeCAUpTscs/usr/bin/FreeCADLink(+0x4ad9) [0x55e2c861bad9]
```
[PcieRiserCase.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8237155/PcieRiserCase.zip)

bambukouk, 2022-03-13:
Yes, they are out of sync as far as OCC? (I think)
I had the same problem (only using latest release, it was OK previously)
Nothing you can do about it afaik until they are the same for Win and Linux (Linux is behind Windows at the moment)
It is undesirable situation and causes problems to those of us who have to work with both OS 
@realthunder hopefully this will be addressed in future release? please.

luzpaz, 2022-03-14:
JFYI, Snap versions use OCC7.6.1 https://snapcraft.io/freecad-realthunder

bambukouk, 2022-03-14:
Sorry forgot about snap (because I don't like it and don't use it 😉)
Waiting (patiently) for appimage....

bambukouk, 2022-04-03:
Latest release for Linux (FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220402-glibc2.12-x86_64.AppImage) still uses 7.5.3, so the (unless you choose to use snap) the issue remains.

I wonder why BTW? Is it to do with Conda?

luzpaz, 2022-04-03:
Yes, it has to do with incompatibility issues looo has been experiencing trying to get OCC 7.6.1 to play well with other dependencies like like pyside2 and who else knows what..it's been rough :disappointed: :

luzpaz, 2022-04-03:
@bambukouk see https://forum.freecadweb.org/viewtopic.php?p=585057#p585057

realthunder, 2022-04-03:
The crash is not directly related to OCC file format problem. Looks like a graphics driver issue, but got somehow triggered by rendering any empty geometry. I can't reproduced it here.

The file format problem has been fixed. Just open the file on Windows LibPack build (i.e. non Conda build) or any other image with OCC 7.6, and save it. No need to recompute, just save. Note that, if your project has multiple files, you'll need to manually save all of them one by one. Save all won't work, as it will only save modified ones. Or, use the following code with your top level document as the active document. After that, the file will be able to open in other non OCC 7.6 releases.

```python
for doc in App.ActiveDocument.getDependentDocuments():
    doc.save()
```

hansfbaier, 2022-04-03:
Thanks! Will try!