realthunder/FreeCAD_assembly3 #1151: Segfault when setting line length (Shift-D) in sketch

ardugitter, 2022-03-13:
In the attached file, if I open Sketch054 and attempt to set the length of the line highlighted in this screenshot, it crashes:

[image]

This is the error text:
```

$ FreeCAD-asm3-Stable-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCArD3BUT
FreeCAD 0.19, Libs: 2022.111R28429 (Git)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bd60) [0x7f22779ded60]
#1  0x7f227754babf in SoPath::append(SoPath const*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/../lib/libCoin.so.80+0x1f
#2  0x7f2279ffd7c5 in Gui::View3DInventorViewer::clearGroupOnTop(bool) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x4b5
#3  0x7f227a000699 in Gui::View3DInventorViewer::checkGroupOnTop(Gui::SelectionChanges const&, bool) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x469
#4  0x7f227a005cdf in Gui::View3DInventorViewer::onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x15f
#5  0x7f227a1ae78a in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x5a
#6  /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so(+0x748dd7) [0x7f2279e61dd7]
#7  0x7f227a1afb95 in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x245
#8  0x7f227a1b0657 in Gui::SelectionSingleton::clearCompleteSelection(bool) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x227
#9  0x7f22231c5112 in finishDistanceConstraint(Gui::Command*, Sketcher::SketchObject*, bool) from /tmp/.mount_FreeCArD3BUT/usr/lib64/SketcherGui.so+0x1f2
#10  0x7f22231c5be1 in CmdSketcherConstrainDistance::activated(int) from /tmp/.mount_FreeCArD3BUT/usr/lib64/SketcherGui.so+0xa41
#11  0x7f2279c74cee in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x3fe
#12  0x7f2277f2ca18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Core.so.5+0x6c0
#13  0x7f2278739928 in QAction::triggered(bool) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Widgets.so.5+0x32
#14  0x7f227873b2b2 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Widgets.so.5+0xc6
#15  0x7f227873ba3b in QAction::event(QEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Widgets.so.5+0xb3
#16  0x7f227873e7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#17  0x7f227874431d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#18  0x7f2279c19d39 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x89
#19  0x7f2277f16606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Core.so.5+0x86
#20  0x7f22782d2b51 in QShortcutMap::dispatchEvent(QKeyEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Gui.so.5+0xd5
#21  0x7f22782d2bc2 in QShortcutMap::tryShortcut(QKeyEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Gui.so.5+0x5c
#22  0x7f22782aaac0 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Gui.so.5+0xb0
#23  0x7f22782b8e4c in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Gui.so.5+0x6c
#24  0x7f22782bb939 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Gui.so.5+0x79
#25  0x7f22782a8c65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Gui.so.5+0x63
#26  /tmp/.mount_FreeCArD3BUT/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7f226ef22010]
#27  /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f227456e76d]
#28  /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/./libglib-2.0.so.0(+0x56a01) [0x7f227456ea01]
#29  /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f227456eaa1]
#30  0x7f2277f450e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Core.so.5+0x5e
#31  0x7f2277f13019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Core.so.5+0x187
#32  0x7f2277f16e8f in QCoreApplication::exec() from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib/libQt5Core.so.5+0xfd
#33  0x7f2279b8c183 in Gui::Application::runApplication() from /tmp/.mount_FreeCArD3BUT/usr/bin/../lib64/libFreeCADGui.so+0x443
#34  /tmp/.mount_FreeCArD3BUT/usr/bin/FreeCADLink(+0x46d5) [0x5618a129c6d5]
#35  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xea) [0x7f22779c9d0a]
#36  /tmp/.mount_FreeCArD3BUT/usr/bin/FreeCADLink(+0x4a6d) [0x5618a129ca6d]
```

BTW, sometimes it crashes just when opening the Sketch itself. Version information:

```
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

Many thanks.

ardugitter, 2022-03-13:
[Problem.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8240452/Problem.zip)

bambukouk, 2022-03-13:
I have no problem either opening the sketch or modifying/adding dimensions to it, but....
once again there is something dodgy about this sketch and it seems that it only works for me because:

```
20:19:47  <App> GeoFeatureGroupExtension.cpp(293): Remove Problem#Sketch054 from Problem#Part because of multiple owner groups
20:19:47  <Exception> GeoFeatureGroupExtension.cpp(338): Auto correct group member for Problem#Part

```

it looks like your setup doesn't do this "auto correct" business, instead just dumps it...

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

ardugitter, 2022-03-13:
> could be to do with the fact that I am using Daily and you are using "Stable" ??

I just downloaded the daily version (same hash as yours) and have exactly the same problem. How come you don't see the problem? Did you click "ok" after changing the line length to 34.5mm?

Also, where does your debug text come from? I don't see any lines starting with a timestamp anywhere.

bambukouk, 2022-03-14:
> I just downloaded the daily version (same hash as yours) and have exactly the same problem. 

don't know 😉
someone tried master branch and he was also all OK?

> How come you don't see the problem? Did you click "ok" after changing the line length to 34.5mm?

just in case 34.5 is significant I will try it again 

> Also, where does your debug text come from? I don't see any lines starting with a timestamp anywhere.

report view - have you toggled it ON ?

bambukouk, 2022-03-14:
[image]

bambukouk, 2022-03-14:
PS, forget this "debug" text...
don't seem to be appearing now
Might be confusion created when I was farting around with your model?

ardugitter, 2022-03-14:
> report view - have you toggled it ON ?

Yes, I have it turned on and there is no output. Here is what I see after I have loaded the problem file:

[image]

I am so confused. Why do I see the problem but not you? :-(

bambukouk, 2022-03-14:
I am not seeing any reports now either, so like I have said - it might have been red herring...?

As to why you are having problems and rest of us don't?
Haven' t got a clue - sorry.

bambukouk, 2022-03-14:
Question:
where do you get `Shift-D` ?
It didn't do anything for me (I don't normally use keyboard shortcuts) so I have checked and the shortcut for vertical dim is `I`

ardugitter, 2022-03-14:
Shift-D sets the length of a line. It is not length in the vertical plane. It is simply the length of the line itself, at any angle.

ardugitter, 2022-03-14:
Here is the icon:
[image]

bambukouk, 2022-03-14:
the shortcut for this one is `K,D` not `Shift,D`  ?

ardugitter, 2022-03-14:
Shift-D -> both keys at the same time. But K,D works for me too.

But I get the same crash by using the icon.

I am sure the crash must happen because we are using different settings. Shall I post mine?

bambukouk, 2022-03-14:
`Shift-D` just doesn't work for me?
Anyway, as you say it doesn't matter for this issue...
Yes, I will compare your settings with mine

ardugitter, 2022-03-15:
I think I must have allocated the Shift-D shortcut manually.

I would prefer not to upload my settings as they contain personal info but I just deleted my .FreeCAD directory and I *still* get the crash.

Please could you try doing "mv .FreeCAD dot_freecad" and see if your setup with blank settings works?

ardugitter, 2022-03-15:
I will close this issue as I have "fixed" the issue by deleting the sketches and starting again.

I don't have the ability to debug because I am using the appimage. Maybe if someone else comes across this thread it will help them.

Thanks for your tips all the same. :-)

bambukouk, 2022-03-15:
You are welcome 👍

realthunder, 2022-03-18:
Should be fixed with the above commit. Note that the problem only occur in `LinkStage3` (i.e. the `Stable`) branch.

closed 2022-03-18 by realthunder

bambukouk, 2022-03-18:
that explains why I didn't see the problem (daily) 👍 
thank you

ardugitter, 2022-03-18:
Thanks for looking into this. I'm glad it's fixed. :-D