realthunder/FreeCAD_assembly3 #1153: [Question] How to translate interface?

Kuzma30, 2022-03-15:
I translate asm3.ts with qt linguist to my language. Then I rewrite this file in Mod directory, but don't see translated messages.  What I need to do?

realthunder, 2022-03-15:
You are supposed to keep the original `asm3.ts`, copy and rename it with your language code, e.g. `asm3_zh-CN.ts`, translate the file, and then [compile](https://wiki.freecadweb.org/Translating_an_external_workbench#Compile_the_translations) it to produce a `qm` file. To compile the translation, you need to install `pyside2-tools` package. If you are using Windows, then you can find `lrelease.exe` in FreeCAD bin directory.

Kuzma30, 2022-03-16:
I create files .ts and qm for my language. What is the next step?

realthunder, 2022-03-16:
That's it. Just start FreeCAD and try.

Kuzma30, 2022-03-16:
Start FreeCAD, still English language in asm3. Another interface is Ukrainian. Can you try this https://github.com/Kuzma30/FreeCAD_assembly3 on your PC?

realthunder, 2022-03-16:
Please rename your ts file to `asm3_uk.ts`. The full language code for Ukrainian is uk-UA. But the postfix is only needed if there are different writing system for the same language.

BTW, while trying your translation, I've discovered two problems. One in the asm3 workbench, the other in my build of FreeCAD. I'll make a new release soon with the fix. Meanwhile, the constraint action title and tooltips should be able to translate. If you are eager to try, you can either manually [change](https://github.com/realthunder/FreeCAD_assembly3/commit/0bfaf710d95fa065ad663cc698965119a638ae65) the asm3 [gui.py](https://github.com/realthunder/FreeCAD_assembly3/blob/master/freecad/asm3/gui.py), or simply clone asm3 github repo to your FreeCAD user directory.

luzpaz, 2022-03-16:
@realthunder have you considered using [crowdin](https://wiki.freecad.org/Translating_an_external_workbench#Send_the_.ts_file_to_a_translation_platform) to manage translations?

Kuzma30, 2022-03-16:
Thank's it works. I will spell my translate and make pull request later.

realthunder, 2022-03-17:
> have you considered using [crowdin](https://wiki.freecad.org/Translating_an_external_workbench#Send_the_.ts_file_to_a_translation_platform) to manage translations?

Not right now I think. Too many things pending.