realthunder/FreeCAD_assembly3 #1154: external geometry problems

bambukouk, 2022-03-16:
I have two sketches in an (active) body.
In one sketch have no problems picking external geometry,
but another sketch refuses to allow me to pick external geometry?
Here an example of first sketch - no problem
[image]
and yet the same active body in another sketch doesn't work:
[image]

I have had temperamental issues with external geometry before but somehow always managed to coax FreeCAD into obeying me :+1: 
but this time for some reason it is NO GO :-1: 
Never had any of these issues before, it seems to be only since the latest 2022.111 release?

I will try to isolate the body in question (original assy file too large and complicated to share) and see if I can attach it.
In the meantime - anybody else with similar experience or any ideas?
Please

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)

```

realthunder, 2022-03-18:
Is it possible to share your file here?

bambukouk, 2022-03-18:
Thank you @realthunder 
I have tried to simplify it and still repeat the problem but haven't made much progress yet.
Happy to email you download link to the whole assembly? if you are willing to have a look at it?

bambukouk, 2022-03-20:
Hi @realthunder 
I have deleted everything un-connected to the issue leaving just a very simple assembly.
Problem is repeatable, here is where you can try it.
(`cylinder_body_LH` active, modifying `sketch_077` and unable to add external geometry from the active body?)
[image]

File:
[ext_geo_issue.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8310933/ext_geo_issue.zip)

realthunder, 2022-04-05:
The image shows you bound on another sketch. Are you trying to pick the geometry from that sketch? I am asking because it looks like you are trying to pick geometry from another feature late in the history, which will cause dependency loop. For some reason, you have disabled auto hiding sketch dependency option. If it's enabled, then those features late in history will be hidden, and for a good reason, because the sketch are not allowed to depend on a feature that depends on it in the first place.

PS. if you want to enable that option, make sure to click `Apply to existing sketch` because those options are saved into the sketch object.

[image]

bambukouk, 2022-04-05:
No, I am trying to modify `Pad108` and working on `Sketch249`
and picking external edge from an earlier feature (in the same body)
It is possible that I am doing something wrong - this problem only occurs when "repairing" old designs
so you might be right about dependency loop?
I will change Sketcher Display settings as per your recommendation and report result
Thank you, as usual for your help :+1:

bambukouk, 2022-04-05:
Sounds like a problem solved, thank you

closed 2022-04-05 by bambukouk