realthunder/FreeCAD_assembly3 #1155: Coordinate system behind menu

HaasNL2, 2022-03-18:
So I was doing some CAM in Link3 and I needed to set the coordinate axis. And I for a while now had this issue that I was forgetting which color is which axis.
And I was thinking that I shouldn't need to remember the colors, and so I loaded 0.2 and notices the small coordinate system in the corner of my view. And Link3 also has this but it is behind the menu.
Is there any way we could make this so it can be moved? Or add the labels to the navigation cube? 

TLDR You sometimes need to know which color is which axis this can be done with the coordinate system in the corner, but in link3 it is behind the menu.

[image]
[image]

bambukouk, 2022-03-22:
:+1:  on that!
I have raised the issue of coordinate axis in bottom RH on "master" forum before:
https://forum.freecadweb.org/viewtopic.php?f=3&t=65143&start=20#p559948
In my case it was Python Console that was getting in a way of coordinate axis.
There is a suggested improvement:
https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:patch-37
but it was beyond my skills to test it (sorry)
Can it possibly be used? in link branch?

HaasNL2, 2022-03-22:
Nice that there is a solution already. I really think it should be in Link 3 as the overlay function pretty much hides it every-time. Although it might be just better to incorporate it into the navigation cube and just scrap the axis completely. As far as I see it both do the same but one shows the X-Y-Z  labels? But I can be wrong.

bambukouk, 2022-03-22:
I (personally) don't use nav cube at all, but would appreciate being able to position coordinate axis somewhere convenient (rather than rigidly fixed in bottom RH corner).
I think we should leave nav cube as is ( simply because you can position it where you want plus you can switch it off if you want..) :+1:

HaasNL2, 2022-03-22:
Couldn't we take the code from the nav cube remove all the extra's until it looks exactly like the coordinate system, but with the position/scale functionality of the nav cube.

bambukouk, 2022-03-22:
I think we should have both `nav cube` and `coordinate system` as long as both can be turned OFF if desired (already possible)
and both can be position as desired (at the moment only possible for nav cube).
Whether and how it can be done I leave to @realthunder  - he always comes up with solutions :+1:

closed 2022-03-28 by realthunder