realthunder/FreeCAD_assembly3 #1156: Problems with CarbonCopy in Sketcher

jpg87Fr, 2022-03-19:
Hello,
I often build from Sketcher WB master sketches that I then use in PartDesign.
In this file, I manage to duplicate with CopyCarbon the 1st of my sketches Sketch042 (Bat1), but not the following ones.
My version of FreeCAD Link Branch is Daily from 2022/01/18 on windows and I also tested with Snap version from 2022/03/07. Not better 
[terrasseEpicV2b.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8309003/terrasseEpicV2b.zip)
!

jpg87Fr, 2022-03-19:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.118.30769 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3b620b713e24c9637db8ea6ad1f1ee88ef13367a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)

OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.307.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.03.07-edge)
Hash: 426e5bafab1f895e6b91f95174b54435be9f43cc
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: French/France (fr_FR)

realthunder, 2022-04-02:
I can't seem to reproduce the problem. Can you please post a screen cast showing your actions?

jpg87Fr, 2022-04-02:
I noticed that if the sketch to copy is not in the active Body you have to press the Ctrl key simultaneously, but in some situations you have to press Ctrl + Alt to detect the sketch to copy.
Here is a short video :

https://user-images.githubusercontent.com/80101333/161392166-40ac57d2-1c76-4cd0-a6a0-e326d86cac72.mp4