realthunder/FreeCAD_assembly3 #1157: key in Sketcher_CreatePolyline

jpg87Fr, 2022-03-19:
I'm also having issues with the M key not working with Sketcher_CreatePolyline.

luzpaz, 2022-03-19:
@jpg87Fr please add you full [about](https://wiki.freecad.org/About) info

jpg87Fr, 2022-03-19:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.118.30769 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3b620b713e24c9637db8ea6ad1f1ee88ef13367a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)

OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.307.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.03.07-edge)
Hash: 426e5bafab1f895e6b91f95174b54435be9f43cc
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: French/France (fr_FR)

ngocminhdao88, 2022-03-20:
Newest version of Sketcher changes all keyboard hotkeys. New hotkey for Sketcher Polyline is GM. To change type line type on the fly you have to press GM again and again.

jpg87Fr, 2022-03-20:
Thank you for the information.
However, it is less practical for the keyboard-unskilled!

closed 2022-03-20 by jpg87Fr