realthunder/FreeCAD_assembly3 #1159: Creating a Document via script generates invalid XML in GuiDocument.xml file

Overruler, 2022-03-19:
Creating a Document from macro or Python Console generates an invalid GuiDocument.xml file.

To reproduce, create a new Document, add a new cube in Part workbench, select the cube and paste the following lines in Python Console:
```

tmp=App.newDocument("test", hidden=True, temp=True)
tmp.moveObject(App.ActiveDocument.copyObject(Gui.Selection.getSelection()[0]))
tmpfile=App.ActiveDocument.getTempFileName("rS")+".FCStd"
tmp.saveCopy(tmpfile)
App.closeDocument(tmp.Name)
tmpfile

```
Unzip the resulting file with 7-Zip. The GuiDocument.xml file contains something like like following:

```
<?xml version='1.0' encoding='utf-8'?>
<!--
 FreeCAD Document, see http://www.freecadweb.org for more information...
-->
<Document SchemaVersion="1" FileVersion="1" Split="0"  <ViewProviderData Count="1">
```

The Document-tag is not closed properly before opening the ViewProviderData-tag.

As a result, the  generated .FCStd file cannot be opened correctly.

luzpaz, 2022-03-20:
please **always** post your full [About](https://wiki.freecad.org/About) info

Overruler, 2022-03-20:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.118.30769 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3b620b713e24c9637db8ea6ad1f1ee88ef13367a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

luzpaz, 2022-03-20:
Can confirm `<Document> tag isn't closed
```
<?xml version='1.0' encoding='utf-8'?>
<!--
 FreeCAD Document, see http://www.freecadweb.org for more information...
-->
<Document SchemaVersion="1" FileVersion="1" Split="0"  <ViewProviderData Count="1">
```
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.307.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.03.07-edge)
Hash: 426e5bafab1f895e6b91f95174b54435be9f43cc
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)
```

closed 2022-03-20 by realthunder