realthunder/FreeCAD_assembly3 #1162: Can we show more debug info for inconsistent constraints?

Kuzma30, 2022-03-24:
In my project I have such error
RuntimeError: Failed to solve (СтанинаСборка#ПластинаДвигателя) СтанинаСборка#Assembly002: inconsistent constraints
Can solver be more informative? Can we show more debug info for inconsistent constraints with it number?

luzpaz, 2022-03-24:
@Kuzma30 please always post your full [About](https://wiki.freecad.org/About) info. And remember to translate back to english errors/warnings that have been translated to other languages

Kuzma30, 2022-03-25:
Add info and full error log

luzpaz, 2022-03-25:
@Kuzma30 Are you using FreeCAD master? Why is your info then not showing Assembly3 version info?

Kuzma30, 2022-03-25:
I use self-builded realthunder Linkdaily branch.

luzpaz, 2022-03-25:
Oh sorry, I missed:
>  Branch: LinkDaily

:facepalm: