realthunder/FreeCAD_assembly3 #1163: [Feature Request] Add "Assembly3 Downgrade" functionality

ToniTen, 2022-03-25:
(Originally posted [in the forum](https://forum.freecadweb.org/viewtopic.php?p=582647#p582647))

For interoperability with people without Assembly3, in a similar way compared to what Fasteners does with "Simplify Shape", some sort of "Assembly3 Downgrade" functionality would be extremely useful. The idea would be to have a simple and quick way to turn Assembly3 assemblies into something that users of FreeCAD that do not have Assembly3 installed can open without any kind of warnings or other strangeness.

(From now on, Part with a capitalized P refers to App::Part while part with no capitalization refers to the contents of an assembly3 assembly that show under the "Parts" section of the assembly)

Executing "Assembly3 Downgrade" on an Assembly3 object (freecad.asm3.assembly.Assembly objects) would convert it to a Part (App:Part) while leaving all the parts at the same location. It would eliminate elements and constraints. It would use the Assembly3 object name as the name of the new Part object. The parts would keep their Placement attribute untouched, but would obviously no longer be constrained in any way.

Additionally, it would be useful to have the option to execute this recursively, to turn an Assembly3 composed of sub-assemblies into an equivalent hierarchy of nested Part objects.

The main goal of this would be interoperability with people without Assembly3 installed, but a side benefit would be that the work of the solver would be reduced since it would not need to work on these components (in a similar fashion to what freezing the assembly does)

realthunder, 2022-03-25:
Assembly3 is usable with upstream FreeCAD, so I don't think it would be of much use for file exchange between FreeCAD. For exchange with other CAD, you can of course use STEP. BTW, importing STEP will in fact use `App::Part` as container by default in my branch. And I have a pending PR doing the same for upstream. Another fun fact, not sure if you know already, is that Assembly3 has buttons for importing STEP using Assembly3 as container.

ToniTen, 2022-03-28:
Sure, you can ask others to install Assembly3, but it's not in the addon manager. Again, look at what Fasteners does (https://wiki.freecadweb.org/Fasteners_Shape) where they have a command to make the fasteners into bog standard FreeCAD shapes, so others opening the model don't need Fasteners (even as Fasteners is much easier to install than Assembly3, since it is available in the addon manager).

As for using STEP as an interchange format, sure, for interfacing with other CAD systems, it works, but it obviously makes everything non-parametric. Also, STEP as a format always causes issues, since it's a huge spec, and AFAIK, most programs only implement a subset of its capabilities (not to mention when they produce invalid geometry straight away). More often than not, models created from STEP files will cause huge performance problems, specially when dealing with TechDraw drawings, assuming they can be projected at all (quite often, IME, STEP models straight from the manufacturer will error out when projecting into 2D and result in empty projections if imported straight away, and thus you need to use Part/Import CAD to import them and manually delete extraneous geometry). It is a known fact that OpenCascade has issues with geometry that isn't 100% good.

I can accept that when interfacing with other CAD, there will be friction, and that STEP is the least bad option, but I personally don't want to have to use it as an exchange format between FreeCAD installations.

I knew about both importing STEP files as App::Part hierarchies, and as freecad.asm3.assembly.Assembly objects, depending on what you want.

luzpaz, 2022-03-28:
> Sure, you can ask others to install Assembly3, but it's not in the addon manager.

@ToniTen indeed, A3 is in the Addon Manager (https://github.com/FreeCAD/FreeCAD-addons/commit/8e1969abbe115deffed90c7070a7ed68cd264613)

ToniTen, 2022-03-29:
> @ToniTen indeed, A3 is in the Addon Manager ([FreeCAD/FreeCAD-addons@8e1969a](https://github.com/FreeCAD/FreeCAD-addons/commit/8e1969abbe115deffed90c7070a7ed68cd264613))

Oh, I didn't know, thanks for the information!