realthunder/FreeCAD_assembly3 #1164: Netgen Bug in newest Version 2022.01.18

harun27, 2022-03-25:
Netgen just doesn't work. It creates a mesh but there isn't any volume. Does anybody else have that problem?

luzpaz, 2022-03-25:
Step-by-step instructions to reproduce please. Also always add you full [About](https://wiki.freecad.org/About) info..

harun27, 2022-03-25:
I've edited it

luzpaz, 2022-03-25:
@harun27 can you recreate in FreeCAD (upstream) master ?

harun27, 2022-03-25:
@luzpaz do you mean with building it myself? If so I haven't done that and don't feel so confident to do so. But I was using this version, where it works just fine:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

I updated today by downloading from the releases https://github.com/realthunder/FreeCAD_assembly3/releases both versions 2022.01.11 and 2022.01.18 but both have the same bug.