realthunder/FreeCAD_assembly3 #1165: AppImage Keeps Crashing In Ubuntu

vyasji-007, 2022-03-27:
OS: Ubuntu 20.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

The latest stable AppImage keeps crashing as soon as creating new file or try to open workbench drop-down or anything in particular on Ubuntu.

I did make sure that the permissions was set to 'Allow executing file as program' checked before starting FreeCAD.
Any idea how to solve this problem?
[image]

bambukouk, 2022-03-27:
Have you tried running FreeCAD from terminal?
also:
https://wiki.freecadweb.org/Debugging

vyasji-007, 2022-03-29:
> Have you tried running FreeCAD from terminal? also: https://wiki.freecadweb.org/Debugging and: https://forum.freecadweb.org/viewtopic.php?f=3&t=30573

No I haven't tried yet, but thanks for pointing out. 

By the way its runs perfectly on Windows though, as well as Linux Mint.

luzpaz, 2022-03-30:
Running from the terminal gives us a clue why it's crashing. Can you please paste the crash error?

susnicek, 2022-04-03:
Dear developers,
at my Debian Sid (Debian 5.16.18-1) box the newest appimage 2022.04.02 also keeps crashing, maybe it is the same problem (the previous version 2022.1.11 behaved the same way for me). The application gives this error output from console:

`./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220402-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCAU7ql7l
FreeCAD 0.20, Libs: 2022.402R26244 +4679 (Git)
libGL error: MESA-LOADER: failed to open nouveau: /usr/lib/dri/nouveau_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open nouveau: /usr/lib/dri/nouveau_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAU7ql7l/AppRun: line 46: 12384 Aborted                 "$PREFIX/bin/$name" "$@"
`

Thaks for help.