realthunder/FreeCAD_assembly3 #1167: 'charmap' codec can't encode ...

jpg87Fr, 2022-03-29:
Hello, here is my configuration :
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.118.30769 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3b620b713e24c9637db8ea6ad1f1ee88ef13367a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)

When I open a new session of FreeCAD Link, I now get the following message :

`09:40:15 'charmap' codec can't encode characters in position 330-342: character maps to <undefined>`

Could someone tell me what I probably deleted by mistake in my configuration ?

bambukouk, 2022-03-29:
If you look at this post on the forum:
https://forum.freecadweb.org/viewtopic.php?f=20&t=25712&start=20#p582484
you will see that @jpg87Fr had exactly the same error message.
He is using the same set up as you.
It is unlikely that this error message is anything to do with your own actions (in my opinion)

BTW I use Linux appimage and didn't get any of these error messages when opening the same file...
@jpg87Fr  tried some things, still with the same error message:
https://forum.freecadweb.org/viewtopic.php?f=20&t=25712&start=10#p582717

realthunder, 2022-03-29:
This is likely a bug in `Start` workbench. You probably opened a file with unicode in file name. If so, the problem is already fixed by [this](https://github.com/realthunder/FreeCAD/commit/fb98757111b2e2f55e0a97fe27dace346c1a3ffd) commit.

bambukouk, 2022-03-29:
well... it was a file with a lot of Cyrillic characters in file names etc.
Opened perfectly fine in Linux but for some reason Win10 got upset about it :-1:

jpg87Fr, 2022-03-29:
I downloaded the StartPage.py file from 18 days ago but now I get this message
`19:03:59  invalid syntax (StartPage.py, line 7)`

realthunder, 2022-03-29:
That's strange. There is no code in line 7. Are you sure you downloaded the raw `py` file? Anyway, you can also try [this](https://drive.google.com/file/d/17wQljo0ztGnPEq6RJoJJRZKUbHkZwf89/view?usp=sharing) Windows build. It's built with Conda, so you may be prompted for installing the solver again.

jpg87Fr, 2022-03-30:
Thank you for your efforts.
I installed the above version and I have this new message (I previously deleted the files from ...AppData>Roaming>FreeCAD) :
`09:05:31  1.641139 <asm3.main> init_gui.py(14): no solver backend found
09:05:33  'utf-8' codec can't decode byte 0xe9 in position 139: invalid continuation byte`

realthunder, 2022-03-30:
Can you please export your user configuration and send it me? `Tools -> Edit parameters... -> Export`

jpg87Fr, 2022-03-30:
here is.
I forgot to mention that I also have a blank start page and that the Preferences settings to indicate the presence of the Samples files and the number of the last used files do not appear either
[jpg87_FCparameters.Zip](https://github.com/realthunder/FreeCAD_assembly3/files/8378068/jpg87_FCparameters.Zip)
.
PS: For sending the file, I replaced the .FcParam extension in .Zip!

realthunder, 2022-03-30:
I can't reproduce the problem. There is no recent files in the config file you posted. And the `Start` workbench loaded okay here. Can you open new file using `File -> Open...`?

realthunder, 2022-03-30:
I just added another possible fix. Can you please download this [StartPage.py](https://github.com/realthunder/FreeCAD/raw/a3a4433718584657db57a48a37e625ce57cf984f/src/Mod/Start/StartPage/StartPage.py) and replace the file in your installation to see if it works?

jpg87Fr, 2022-03-30:
I replaced the StartPage.py file with yours and I still have this message (as well as a blank page)
`19:03:54  'utf-8' codec can't decode byte 0xe9 in position 138: invalid continuation byte`
I put my old config files back and here is the corresponding settings file
[jpg87_FCparameters2.Zip](https://github.com/realthunder/FreeCAD_assembly3/files/8382682/jpg87_FCparameters2.Zip)

I am attaching a file that I have been working on for the last few days and on which upon reopening in the latest version (conda) that you indicated to me a few hours ago I have some problems with the Datums Planes (I will take it back and fix the mistakes, so if you don't have time to watch it that's no problem for me).
[terrasseEpicV2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8382745/terrasseEpicV2.zip)

realthunder, 2022-03-31:
Can you please try [this](https://raw.githubusercontent.com/realthunder/FreeCAD/8e2c6d6753d6cdf65d7e9c3cc13bf828566490c8/src/Mod/Start/StartPage/StartPage.py) StartPage?

Also, can you please zip and send me your directory `C:\Users\<your_username>\AppData\Local\Temp\FreeCADStartThumbnails`?

jpg87Fr, 2022-03-31:
I deleted my starting configuration and replaced StartPage.py with the new one. 
Always the same message (only the position has changed) and a blank page. 
I'm sorry to cause you this trouble.
`09:10:40  'utf-8' codec can't decode byte 0xe9 in position 139: invalid continuation byte
`

realthunder, 2022-03-31:
No problem. It's always good to get chance for a bug fix.

I think the problem is in the legacy thumbnail cache files. Can you please send me that directory? But the strange thing is you even see the blank page when you delete `link.user.cfg`?

jpg87Fr, 2022-03-31:
Thousand excuses ! I just noticed that I had changed StartPage.py in the stable version and opened daily! Everything works normally in both versions.
Thank you very much for your efforts and your patience (and all your work for FreeCAD).

realthunder, 2022-03-31:
Great that it all worked out! Probably the last change is not needed.

jpg87Fr, 2022-03-31:
I just tested also the (non-conda) version FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220118 and it also works again.

closed 2022-03-31 by jpg87Fr