realthunder/FreeCAD_assembly3 #1171: Angle problem

hamza00789, 2022-03-30:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.118.30769 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3b620b713e24c9637db8ea6ad1f1ee88ef13367a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

There is no difference between Sketch and Sketch 001. I extrude the sketch and used an angle. No problem, and it worked just fine. when I extrude Sketch 001. But when I tried to angle the extude001, the angle didn't work. The two sketches are the same. Is there a difference between the two sketches or am I making a mistake?  I often have this problem. One thing I want to say here is that I had this problem not only in this new version but also in the old version. 

[File.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8383561/File.zip)

bambukouk, 2022-03-31:
[File(bambuko).FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8387263/File.bambuko.FCStd.zip)

:+1:

hamza00789, 2022-03-31:
> [File(bambuko).FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8387263/File.bambuko.FCStd.zip)
> 
> 👍 when copying the sketch, copy just a sketch not a sketch plus coordinates

Did I mention that I copied the sketch? I did not copy the sketch. I have recreated the sketch001. I recreated it so that I could explain the problem that it does this way sometimes.

bambukouk, 2022-03-31:
You didn't say that you have re-created it instead of copying either...
So... if you say it happens "sometimes" than you don't need two sketches?
If you try to extrude a sketch enough times than eventually (sometime) it will fail?
I will have to have a look at your re-created sketch to see if there is any difference indeed :+1:  ?

bambukouk, 2022-03-31:
I am not doubting you :-)
just wasn't clear...
There is something odd because I also had (reliably) failure of angled extrusion, but only if I copied coordinate system as well as the sketch, and this (I think) is the clue to what the problem is (i.e. attachment of the sketch - it looks fine on the screen but causes problems when using angled extrusion)
Give me sometime and I will experiment more

hamza00789, 2022-03-31:
This error message comes when it problem.
[1.txt](https://github.com/realthunder/FreeCAD_assembly3/files/8389383/1.txt)

bambukouk, 2022-03-31:
yes, same here...

hamza00789, 2022-03-31:
Also, I want to send two videos. The problem is explained in these videos. But the video is too big.

bambukouk, 2022-03-31:
well... I wouldn't worry about the videos
it seems clear (to me at least) what the problem is - it's the solution that we are after :+1:

bambukouk, 2022-03-31:
There seem to be long history of problem with extrude taper angle
apparently OCC related?
https://duckduckgo.com/?q=part+extrude+angle&sa=Search&sites=forum.freecadweb.org&atb=v297-1&ia=web

hamza00789, 2022-04-14:
https://drive.google.com/file/d/144MQOzygs1o8Bi6ZQ6j86RHxCrt9mlf4/view?usp=sharing 
 
There are two videos. In both videos, the sketch is created differently. In the first video comes the error message from the way the sketch is created.   But in the second video, the way the sketch is created, the error message does not come.

hamza00789, 2022-05-11:
https://user-images.githubusercontent.com/92737149/167801816-95453990-29f5-4068-81bb-a56f0b524060.mp4