realthunder/FreeCAD_assembly3 #1172: Object remains visible after visibility is turned off for all objects

jx2014, 2022-04-01:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

Just as screenshot is showing, the object remains visible even when visibility is turned off for all:
![image](https://user-image
[FenceBirdFeeder.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8394812/FenceBirdFeeder.zip)
s.githubusercontent.com/7607066/161197931-4e2284f6-a219-48fb-9616-661f07308743.png)

bambukouk, 2022-04-01:
Alt+click :+1: 
[image]

jx2014, 2022-04-02:
Thanks, I tried it, and it worked. How did it work? I had to press alt + click to turn off the object. Is this a bug?

bambukouk, 2022-04-02:
No, it's not a bug.
you `click` to toggle `visibility`
you `alt+click` to toggle `show on top` 
you can close this "issue" :+1:

closed 2022-04-02 by luzpaz