realthunder/FreeCAD_assembly3 #1173: No transparent overlay TreeView and very wired file browsing.

QuakeString, 2022-04-02:
Overlay treeview is not transparent and the file browsing is very absurd. 

OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.329.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.03.30-edge)
Hash: fe291883c37a2f70b67183aaccbca61e4abf5295
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)

here is video 
https://drive.google.com/file/d/1EmRrXyRCvLKXjsNdQnFUT8rTIwmE5A5M/view?usp=sharing

realthunder, 2022-04-02:
Are you using Wayland? Is so, try [this](https://github.com/ppd/freecad-ppd/issues/23#issuecomment-1049097902).

QuakeString, 2022-04-02:
I'm using X11.
[image]

oficinerobotica, 2022-04-03:
I can confirm both the lack of transparency for the overlay and the missing icons in the file browser dialog using the snap build.
 
In both the appimage and snap , the UI becomes really slow to respond in some cases. There can be up to 3/4 seconds delay between click and result of the click. At times, even if the model is not involved(ie click ok on the preferences dialog) I got an  app freeze using snap. With the appimage it only takes a really long time. 
This behavior is not at all consistent, I will try to catch it and give more details.

P.S thank you so much for the new release :)

OS: KDE neon User - 5.24 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.402.26244 +4679 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 2ed36c0a234b3871a790ac8437f496f29e025ded
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

oficinerobotica, 2022-04-03:
Notice how long it takes between click and action especially when dialogs need to appear. Again, this is not consistent. For example, expanding items in the tree can be slow and at the next click immediate.

https://user-images.githubusercontent.com/26909793/161411656-729265ae-c023-43d4-bcef-7869bb0f00da.mp4

realthunder, 2022-04-03:
Is the UI slowness also presented in AppImage?

oficinerobotica, 2022-04-03:
...AppImage...
Yes. The video above is recorded with the latest appimage.

...Snap...
With the snap is more severe.
I don't use the snap because it is almost impossible to browse the file system from the FC dialog as seen from the OP video. Appart from the lack of icons in the file browser dialog for navigating  the file system, the drop down dialogs are unresponsive to clicks making it impossible to select anything.

realthunder, 2022-04-03:
The tree view slow response should be fixed by the above commit. But the transparency and file browser icon seems to be either snap or (KDE + snap) related. @ppd have you encounter any problem like this?

bambukouk, 2022-04-03:
BTW  (in case it helps) - I have not noticed anything like this 

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit

Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9

or/either

Version: 2022.402.26244 +4679 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 2ed36c0a234b3871a790ac8437f496f29e025ded

Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

ppd, 2022-04-03:
I can't download the first video: "We can’t connect to the server at drive.google.com.".

I have no issues with the file dialog. All the icons are there. @luzpaz uses Manjaro and could test this weirdness, just in case it's related to some oddity wrt the snapd implementation on Manjaro.

ppd, 2022-04-03:
Ah, now the video works. I'm not seeing such weirdness. KDE has a global menu, right? Maybe there's some strange interaction with it; obviously I'm just speculating.

QuakeString, 2022-04-03:
I'm just trying the new release daily build AppImage, and the file browser issue fixed, Settings page takes too long time in previous Snap build but thats fixed in latest daily build AppImage. also tree view transparency got fixed.
But new issue arise, that is there nothing on start page. absolutely blank.
[image]

oficinerobotica, 2022-04-03:
@ppd I am trying to give it another spin at the snap package but when I try to install it I get this error
```
 sudo snap install freecad-realthunder --edge
[sudo] password di : 
error: cannot perform the following tasks:
- Run configure hook of "freecad-realthunder" snap if present (run hook "configure": cannot locate base snap core20: No such file or directory) 
```

but listing snaps I get
```
snap list --all
Name                                  Version          Rev    Tracking       Publisher         Notes
bare                                  1.0              5      latest/stable  canonical✓        base
core                                  16-2.54.4        12821  latest/stable  canonical✓        core
core18                                20220309         2344   latest/stable  canonical✓        base
core20                                20220318         1405   latest/stable  canonical✓        base
gnome-3-38-2004                       0+git.1f9014a    99     latest/stable  canonical✓        -
gtk-common-themes                     0.1-59-g7bca6ae  1519   latest/stable  canonical✓        -
kde-frameworks-5-91-qt-5-15-3-core20  5.91.0           1      latest/stable  kde✓              -
kde-frameworks-5-core18               5.61.0           32     latest/stable  kde✓              -
kde-frameworks-5-qt-5-15-3-core20     5.87.0           8      latest/stable  kde✓              -
telegram-desktop                      3.4.3            3544   latest/stable  telegram.desktop  -
```

How can I overcome this to report back on the op's issue?

ppd, 2022-04-03:
@oficinerobotica Oh, hm, I've never seen that error before. Do you have `/snap/core20` on your system?
E.g.
```shell
$ ls -l /snap/core20/
total 0
drwxr-xr-x 21 root root 348 Mär  4 08:45 1376
drwxr-xr-x 21 root root 348 Mär 18 06:47 1405
lrwxrwxrwx  1 root root   4 Mär 30 09:09 current -> 1405
```

Did you move `/snap` for some reason?

realthunder, 2022-04-03:
@QuakeString Can you please find the directory `/tmp/FreeCADStartThumbnails`, zip it and send to me? Also, please right click the blank Start page and choose `View source`, and copy and send me the HTML source of the start page. Thanks.

QuakeString, 2022-04-04:
@realthunder right click on blank start page does not bring up the right click menu. 
attached /tmp/FreeCADStartThumbnails folder content below.
[FreeCADStartThumbnails.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8406202/FreeCADStartThumbnails.zip)

realthunder, 2022-04-04:
Hmm, can't find anything wrong there. Are there any error in Report view with the blank start page?

QuakeString, 2022-04-04:
Report View also empty. there is no entry.

Taack, 2022-04-04:
@QuakeString I have had a similar issue with the AppImage, the errors printed when launching from the command line helped me to find a solution.

This comment solved the issue for me (removing usr/lib/libstdc++.so.6 and usr/lib/libstdc++.so from the AppImage in order to use the system wide libstdc):
https://forum.freecadweb.org/viewtopic.php?p=510504&sid=15a6fda0f651398024fe6491c364114c#p510504

This worked for me on 
`FreeCAD_weekly-builds-28611-Linux-Conda_glibc2.12-x86_64.AppImage `

Hope this helps..

QuakeString, 2022-04-04:
I don't see any issue output on command line.
[image]

Taack, 2022-04-04:
FYI `FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220402-glibc2.12-x86_64.AppImage` crashes during startup with the following logs (on Mageia 9):

```
$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220402-glibc2.12-x86_64.AppImage
/tmp/.mount_FreeCAc46NNt
FreeCAD 0.20, Libs: 2022.402R26244 +4679 (Git)
During initialization the error "[Errno 13] Permission denied: ''" occurred in /home/auo/.FreeCAD/Mod/CADExchanger/Init.py
Please look into the log file for further information
Assembly4 workbench (v0.11.10) loaded
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_FreeCAc46NNt/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib64/libLLVM-14.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_FreeCAc46NNt/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib64/libLLVM-14.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAc46NNt/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib64/libLLVM-14.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: swrast
1.213395 <asm3.main> init_gui.py(14): no solver backend found
During initialization the error "No module named 'osecore'" occurred in freecad.ose3dprinter
--------------------------------------------------------------------------------
[ . . . ]
--------------------------------------------------------------------------------
/tmp/.mount_FreeCAc46NNt/AppRun : ligne 46 : 1145218 Abandon                 (core dumped)"$PREFIX/bin/$name" "$@"
```

Doing procedure described above allow me to start FreeCAD-asm3.. But now I am facing the same issue than QuakeString.

```
$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220402-glibc2.12-x86_64.AppImage  --appimage-extract
$ cd squashfs-root/
$ rm ./usr/lib/libstdc++.so*
$ ./appimagetool-x86_64.AppImage squashfs-root
$ ./FreeCAD_Link_Branch-x86_64.AppImage
```

realthunder, 2022-04-04:
@Taack Looks like the same issue [here](https://github.com/FreeCAD/FreeCAD-Bundle/issues/67). Please try the workaround mentioned there.

Taack, 2022-04-04:
@realthunder thanks for the tip, I do not know we were able to interpose symbols for an AppImage. 

FYI, when launching `FreeCAD-asm3` from `squashfs-root`, I do obtain this error:
```
$ ./usr/bin/FreeCADLink
FreeCAD 0.20, Libs: 2022.402R26244 +4679 (Git)
Fontconfig error: Cannot load default config file: No such file: (null)
```

Looks like a `QtWebEngineProcess` error. I will try to look at it further and compare `strace` with my working version and tell you if I do find anything.

bambukouk, 2022-04-05:
Is this commit likely to address slow UI display issues as described by @oficinerobotica  ?
or is it only specific to error messages reported later in this issue?
(sounds like it only is `tree view icon refresh`?)

I have edited my original post in this issue (https://github.com/realthunder/FreeCAD_assembly3/issues/1173#issuecomment-1086860564) to reflect my later experiences when using more complicated assembly.

On commanding zoom or pan nothing happens for a while and then instead of smooth progress display jumps to a new position/zoom.

I have tried changing rendering cache away from experimental but it didn't make difference?

I have reverted back to my previous appimage (FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage) and all is back to normal...

or should I raise `separate issue`?
[image]

bambukouk, 2022-04-05:
Just tested - solved what was bothering me :+1: 
Thank you!!

Piezoid, 2022-10-09:
I have the same issue on Arch Linux with `freecad-linkstage3-git` built from AUR. Both the overlays transparency and file dialog icons are missing.

It does indeed look like an interaction with KDE session settings: doing `unset XDG_CURRENT_DESKTOP KDE_FULL_SESSION` fixed it.

<details>
<summary>ldd output on my system:</summary>

```
        linux-vdso.so.1 (0x00007fff409b5000)
        libFreeCADGui.so => /usr/lib/freecad/lib/libFreeCADGui.so (0x00007f9e47000000)
        libFreeCADApp.so => /usr/lib/freecad/lib/libFreeCADApp.so (0x00007f9e46800000)
        libQt5Widgets.so.5 => /usr/lib/libQt5Widgets.so.5 (0x00007f9e46000000)
        libQt5Gui.so.5 => /usr/lib/libQt5Gui.so.5 (0x00007f9e45800000)
        libQt5Core.so.5 => /usr/lib/libQt5Core.so.5 (0x00007f9e45200000)
        libFreeCADBase.so => /usr/lib/freecad/lib/libFreeCADBase.so (0x00007f9e44e00000)
        libpython3.10.so.1.0 => /usr/lib/libpython3.10.so.1.0 (0x00007f9e44a00000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f9e44600000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f9e48470000)
        libc.so.6 => /usr/lib/libc.so.6 (0x00007f9e44419000)
        libCoin.so.80 => /usr/lib/libCoin.so.80 (0x00007f9e43a00000)
        libGL.so.1 => /usr/lib/libGL.so.1 (0x00007f9e46f7a000)
        libQt5PrintSupport.so.5 => /usr/lib/libQt5PrintSupport.so.5 (0x00007f9e46f0a000)
        libQt5Svg.so.5 => /usr/lib/libQt5Svg.so.5 (0x00007f9e46eb1000)
        libQt5Network.so.5 => /usr/lib/libQt5Network.so.5 (0x00007f9e45e8f000)
        libspnav.so.0 => /usr/lib/libspnav.so.0 (0x00007f9e48464000)
        libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 => /usr/lib/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 (0x00007f9e467d0000)
        libFreeCADRenderer.so => /usr/lib/freecad/lib/libFreeCADRenderer.so (0x00007f9e4845e000)
        libQt5Xml.so.5 => /usr/lib/libQt5Xml.so.5 (0x00007f9e4678e000)
        libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 => /usr/lib/libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 (0x00007f9e4673d000)
        libboost_filesystem.so.1.80.0 => /usr/lib/libboost_filesystem.so.1.80.0 (0x00007f9e46719000)
        libxerces-c-3.2.so => /usr/lib/libxerces-c-3.2.so (0x00007f9e43600000)
        libm.so.6 => /usr/lib/libm.so.6 (0x00007f9e45118000)
        /lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f9e484e2000)
        libboost_program_options.so.1.80.0 => /usr/lib/libboost_program_options.so.1.80.0 (0x00007f9e466d2000)
        libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007f9e457c7000)
        libz.so.1 => /usr/lib/libz.so.1 (0x00007f9e45e75000)
        libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007f9e4502d000)
        libmd4c.so.0 => /usr/lib/libmd4c.so.0 (0x00007f9e466bc000)
        libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007f9e44936000)
        libdouble-conversion.so.3 => /usr/lib/libdouble-conversion.so.3 (0x00007f9e457b1000)
        libicui18n.so.71 => /usr/lib/libicui18n.so.71 (0x00007f9e43200000)
        libicuuc.so.71 => /usr/lib/libicuuc.so.71 (0x00007f9e43001000)
        libpcre2-16.so.0 => /usr/lib/libpcre2-16.so.0 (0x00007f9e45723000)
        libzstd.so.1 => /usr/lib/libzstd.so.1 (0x00007f9e4488d000)
        libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007f9e42ec4000)
        libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f9e42d81000)
        libexpat.so.1 => /usr/lib/libexpat.so.1 (0x00007f9e44dd5000)
        libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007f9e4485b000)
        libOpenGL.so.0 => /usr/lib/libOpenGL.so.0 (0x00007f9e443ee000)
        libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007f9e44336000)
        libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2 (0x00007f9e439ac000)
        libssl.so.1.1 => /usr/lib/libssl.so.1.1 (0x00007f9e4356b000)
        libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0x00007f9e42a00000)
        libQt5Qml.so.5 => /usr/lib/libQt5Qml.so.5 (0x00007f9e42400000)
        libboost_atomic.so.1.80.0 => /usr/lib/libboost_atomic.so.1.80.0 (0x00007f9e45e6b000)
        libnsl.so.3 => /usr/lib/libnsl.so.3 (0x00007f9e44dcc000)
        libcurl.so.4 => /usr/lib/libcurl.so.4 (0x00007f9e42958000)
        libicudata.so.71 => /usr/lib/libicudata.so.71 (0x00007f9e40600000)
        libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f9e42889000)
        libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007f9e44839000)
        libcap.so.2 => /usr/lib/libcap.so.2 (0x00007f9e44dc0000)
        libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f9e404b7000)
        liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007f9e4430c000)
        liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f9e4398a000)
        libpcre2-8.so.0 => /usr/lib/libpcre2-8.so.0 (0x00007f9e42ce6000)
        libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007f9e43540000)
        libkrb5.so.3 => /usr/lib/libkrb5.so.3 (0x00007f9e42328000)
        libk5crypto.so.3 => /usr/lib/libk5crypto.so.3 (0x00007f9e4285b000)
        libcom_err.so.2 => /usr/lib/libcom_err.so.2 (0x00007f9e45027000)
        libkrb5support.so.0 => /usr/lib/libkrb5support.so.0 (0x00007f9e4397c000)
        libkeyutils.so.1 => /usr/lib/libkeyutils.so.1 (0x00007f9e44db9000)
        libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f9e4396a000)
        libtirpc.so.3 => /usr/lib/libtirpc.so.3 (0x00007f9e4282e000)
        libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x00007f9e4048b000)
        libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007f9e4280c000)
[... cut ...]
        libbrotlidec.so.1 => /usr/lib/libbrotlidec.so.1 (0x00007f9e43532000)
        libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f9e40437000)
        libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f9e40411000)
        libXau.so.6 => /usr/lib/libXau.so.6 (0x00007f9e44307000)
        libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007f9e4230d000)
        libunistring.so.2 => /usr/lib/libunistring.so.2 (0x00007f9e40265000)
        libbrotlicommon.so.1 => /usr/lib/libbrotlicommon.so.1 (0x00007f9e40242000)
```
</details>

I'm not using Wayland.

Linny3D, 2023-01-30:
Maybe as additional information regarding the transparency problem and the slow Settings dialog:

I have the same problem, but not only with snap or appimage, but also with a self compiled version from the github repo.
While using the `unset XDG_CURRENT_DESKTOP KDE_FULL_SESSION` trick, mentioned by @Piezoid works with the snap build, it doesn't help with the self compiled version. 

The Transparency problem is linked to KDE as it works flawlessly on Gnome and IceWM, but the problem with the slow/non-responsive settings dialog persists in all Desktop Environments.

Unfortunately i'm not familiar enough with C++ to be of any more help here.

Side note:

I'm using X11 not Wayland