realthunder/FreeCAD_assembly3 #1174: Path Array-related Issues

hamza00789, 2022-04-02:
The transform option is missing. 
  
[image] 

There is another problem. When an object created in part design is tried to run on the path, it does not move on the path. I tried to run revolution on a path. The result did not turn out as expected.  

[image] 

file               

[Revolution.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8402203/Revolution.zip) 

But when I tried to run the cone on the path, the result was as expected. 

[image] 

file 

[Cone.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8402205/Cone.zip)

realthunder, 2022-04-04:
The context menu is fixed with the above commit.

The offset problem is not really a bug.  Because the way your sketch is created, the wire is internal offset to the origin. Draw your sketch around the origin shall fix the problem.

hamza00789, 2022-04-04:
> 

I don't understand what you mean. Can you explain further?

closed 2022-04-04 by hamza00789