realthunder/FreeCAD_assembly3 #1176: Problem with configuration tables

jpg87Fr, 2022-04-03:
Following a Kuzma30 problem stated on the forum (https://forum.freecadweb.org/viewtopic.php?f=20&t=25712&p=585711#p585383), I am attaching a video to show the failures :
https://www.dropbox.com/s/rpj3nxac0yy0pl1/PbTableConfig.mp4?dl=0

jpg87Fr, 2022-04-04:
I modified the assembly by creating a sub-assembly (bottom image).
The main assembly (top image) reacts to the configuration change, but the sub-assembly requires one and even several successive recalculations before positioning itself correctly, which also generates the update of the main assembly.
The spreadsheet page still not updating correctly.

[04-03essai.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8409815/04-03essai.zip)
(7z archive)
[image]
[image]
`OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.402.26244 +4679 (Git)
Build type: Release
Branch: LinkDaily
Hash: 2ed36c0a234b3871a790ac8437f496f29e025ded
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)
`

realthunder, 2022-04-05:
The Spreadsheet not updating problem is fixed by the above commit.

The sub-assembly requiring manual recompute is not really a bug actually. You can trigger recompute after change the configuration combo box by pressing return key. This is by design, which arguably could use some improvement. It is mainly to allow smoother changing of the property value, so that a full recompute is only performed after user confirms it by either pressing return, or change widget focus. 

The fasten object renaming is a problem with the [fastener](https://github.com/shaise/FreeCAD_FastenersWB) workbench. You might want to report it there.

I don't seem to experience any panic in constraint, which also didn't seem to show in the video you post. Can you please elaborate?

jpg87Fr, 2022-04-05:
Thank you for your response.
In fact, I had recreated the constraint elements before making my video, so they no longer appeared. To see it, you have to refer to the files of Kusma30 (the author) :
https://github.com/Kuzma30/FREECAD_CNC/tree/main/%D0%A7%D0%B0%D1%81%D1%82%D0%B8%D0%A1%D1%82%D0%B0%D0%BD%D0%BA%D0%B0/%D0%A1%D1%82%D0%B0%D0%BD%D0%B8%D0%BD%D0%B0

jpg87Fr, 2022-04-05:
I tested my file with the daily version of April 05 : the instances of Link(sAsПосадка) have a problem.
I remade the file by putting instead 2 separate links of sAsПосадка and it works again as described previously (by recalculating sAsПосадка).
Files that worked with the April 04 version and no longer work with the April 05 version :
https://www.dropbox.com/s/518f4lx7618pkl6/forTestWithSubAss.zip?dl=0

The new files :
https://www.dropbox.com/s/pu13xjbfw14vp4i/forTestWithSubAss2.zip?dl=0

realthunder, 2022-04-06:
It's caused by the same problem in #1182. 04.06 image has just been release. It should fix the problem.

jpg87Fr, 2022-04-06:
Thank you very much for the speed of your response. Everything seems to work.

closed 2022-04-06 by jpg87Fr