realthunder/FreeCAD_assembly3 #1177: Unable to fetch addons

horfee, 2022-04-04:
When trying to display the list of addons, an error is triggered.

Step to reproduce :
In the toolbar, select tools > add-on manager
The window is displayed, but nothing happen.

Report is:

12:51:02  Traceback (most recent call last):
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 980, in send
    self.connect()
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1454, in connect
    self.sock = self._context.wrap_socket(self.sock,
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/ssl.py", line 1040, in _create
    self.do_handshake()
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
12:51:02  ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)
12:51:02  
During handling of the above exception, another exception occurred:

12:51:02  Traceback (most recent call last):
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
12:51:02  urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>
12:51:02  Debug: addon_flags.json not found
12:51:02  Traceback (most recent call last):
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 980, in send
    self.connect()
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/http/client.py", line 1454, in connect
    self.sock = self._context.wrap_socket(self.sock,
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/ssl.py", line 1040, in _create
    self.do_handshake()
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
12:51:02  ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)
12:51:02  
During handling of the above exception, another exception occurred:

12:51:02  Traceback (most recent call last):
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
12:51:02    File "/Users/xxx/Applications/FreeCAD.app/Contents/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
12:51:02  urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>
12:51:02

horfee, 2022-04-04:
Version is 
OS: macOS 12.3
Word size of FreeCAD: 64-bit
Version: 2022.402.26244 +4679 (Git)
Build type: Release
Branch: LinkDaily
Hash: 2ed36c0a234b3871a790ac8437f496f29e025ded
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

horfee, 2022-04-04:
No it's not a crash of the app, just unable to install any workbench... It seems a missing CA certificate, or a trusted CA not well setup...

luzpaz, 2022-04-07:
Can't reproduce on
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.406.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.06-edge)
Hash: 0c8e437ef38cdc5dd087529ce6825c6a2a130868
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)
```

realthunder, 2022-04-08:
Can you try again with the latest BigSur release? I've made some changes.

horfee, 2022-04-10:
Hello, it is working now ! Thank you very much !

closed 2022-04-10 by horfee