realthunder/FreeCAD_assembly3 #1178: Dependent objects to selection

hamza00789, 2022-04-04:
This option is not available in Link Branch. Can I request this at Link Branch? I need it often. Could this feature be available within Link Branch in the future? 

[image]

hamza00789, 2022-04-04:
> 
This option is used if I want to see or hold the children of a parent. This feature was not previously available in freecad. Which was quite annoying for me. It is most commonly used when the file is large enough. For example, I used the fill command. Now Fill is a father and has many children. the children may be up and down in a tree. It's hard to find these children in a large tree.

realthunder, 2022-04-05:
But that command will select all children and their children as well. Do you find that useful? I saw this function is added in upstream recently, but having doubt how useful would that be. What you require is to select only the direct children I guess. BTW, for your example, the Fill object, it has a property listing all the objects used, just click that property in the property view will select all those objects.

hamza00789, 2022-04-05:
This command is very useful. I tried It only selected object children, not their children. You are talking about fill property listing. Fill property listing is useful but this command is more useful. I wanted to give you some advice in this regard. If some options are added to the property listing, it will be better than this order. One eye icon.filter object-dependent feature.  

[image]

hamza00789, 2022-04-05:
This command is very useful. I tried It only selected object children, not their children.

hamza00789, 2022-04-06:
Is this problem solved? Should I close it?

realthunder, 2022-04-06:
Oops, wrong commit message, which is supposed to close this issue automatically. The new commands are added to 'Context menu -> Selection -> Select direct dependent objects'

closed 2022-04-06 by realthunder