realthunder/FreeCAD_assembly3 #1182: Release 2022.05.04 breaks assemblies.

oficinerobotica, 2022-04-05:
Please guide me on how to give more info on this bug because I don't know what to look for. I get this result by opening the same file, no intermediate saves with diferent releases.
This is the assembly with release 
OS: KDE neon User - 5.24 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.402.26244 +4679 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 2ed36c0a234b3871a790ac8437f496f29e025ded
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
[image]

and with release:
OS: KDE neon User - 5.24 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.405.26244 +4689 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 56c59284784b03af911a0c7121410e84de6d50a2
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
[image]

closed 2022-04-06 by realthunder

bambukouk, 2022-04-06:
does it break the model irreversibly?
i.e. should I stop using 2022.0405 until the commit is released?

oficinerobotica, 2022-04-06:
@bambukouk realthunder has been lightning fast as usual and it seems the bug has been corrected. 

Don't save your file under the 05.04 release.  If you open it with a previous release the file is fine.

bambukouk, 2022-04-06:
Indeed he has :+1:  been as fast as usual - thank you!

realthunder, 2022-04-06:
04.06 release is available. Please try.