realthunder/FreeCAD_assembly3 #1184: The shortcut key is not working (note: Outside the draft workbench)

hamza00789, 2022-04-06:
The move to group key (Shift+G) only works in the draft workbench. In the old version, this key also worked outside the draft workbench. Previously, this key worked in all workbenches.

realthunder, 2022-04-06:
This can actually be considered as a bug that's been fixed in recent release. The workbench specific command and its shortcut is only supposed to be active if it is visible in either toolbar or menu bar. This is so that user can assign the same shortcut to different command in different workbench. If you really want some command to be available globally, you can create your own global toolbar with that command.

[image]

hamza00789, 2022-04-06:
Thanks for the quick reply.

closed 2022-04-06 by hamza00789