realthunder/FreeCAD_assembly3 #1185: [BUG] Color information is lost on frozen assemblies from partially loaded files inserted into frozen assemblies

ToniTen, 2022-04-06:
Related to #1180 
### Bug Description
When an assembly that is composed of frozen assemblies that come from partially loaded files (via links) is itself frozen, color information on the frozen assemblies is lost
### Steps to reproduce
1. Extract the files in [this archive](https://github.com/realthunder/FreeCAD_assembly3/files/8425540/color_test2.zip) into a directory 
2. Open color_test.FCStd
3. Observe that FullAssembly contains 4 Parts: CylinderAssembly (Green, frozen, linked from parts.FCStd), CubeAssembly (Red, not frozen, linked from parts.FCStd), SphereAssembly (Blue, frozen, local to color_test.FCStd), and Tube (Yellow, not an assembly, local to color_test.FCStd) 
[image]
4. Observe that parts.FCStd is partially loaded, because CylinderAssembly is a frozen assembly linked from parts.FCStd
5. Freeze FullAssembly
6. Observe that all parts keep their color, save for CylinderAssembly 
[image]

FreeCAD version information
```
OS: Ubuntu Core 20 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.406.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.06-edge)
Hash: 0c8e437ef38cdc5dd087529ce6825c6a2a130868
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: Spanish/Spain (es_ES)
```

bambukouk, 2022-04-06:
see also similar problems in the past:
https://github.com/realthunder/FreeCAD_assembly3/issues/590#issue-873620329
https://github.com/realthunder/FreeCAD_assembly3/issues/606#issue-882163435

realthunder, 2022-04-06:
I couldn't reproduce the problem. Did you install Assembly3 workbench manually by cloning the repository in your user directory, or through add-on manager, maybe? If so, please manually sync the repo, or simply delete it. The prebuilt image preinstalled the latest Assembly3 workbench, but will opt to use the one in the user directory if found.

ToniTen, 2022-04-06:
I'm using the snap version, and that should use a different .FreeCAD directory
[image]

I do have Assembly3 installed via the addon manager for the standalone FreeCAD daily build from the Ubuntu PPA

ToniTen, 2022-04-06:
I also updated Assembly3 on my FreeCAD daily build from the Ubuntu PPA, with the same files, and all color information was lost when freezing the assembly
[image]
[image]

FreeCAD version information for the other FreeCAD install
```
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 6eafd180306cf19028d361115a47cac2fbee55a3
Python 3.6.9, Qt 5.9.5, Coin 4.0.0a, OCC 7.5.2
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * fasteners 0.11.2
  * Assembly3 0.11.2
  * fcgear 1.0.0
  * Manipulator 1.4.3
```

ToniTen, 2022-04-06:
By the way, #1180 is properly fixed in the latest snap, this is a related but different issue

closed 2022-04-07 by realthunder

fish1968, 2024-09-03:
Is this bug fixed or not? I used 0.21.2 in Windows, still exist.

luzpaz, 2024-09-03:
@fish1968 most likely fixed in development 0.22.x (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

fish1968, 2024-09-04:
> @fish1968 most likely fixed in development 0.22.x (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

It works! Thank you so much. 

Export in AP203, or AP214 International Standards both work well.