realthunder/FreeCAD_assembly3 #1188: Plugin Modern UI crash

jean-raphael, 2022-04-06:
a this point this is a little detail compared to the huge improvements that this version add to freecad

but Modern Ui does not work on this , for example if i choose modern ui tabs , then settings, if i try to move workbench freecad crash

and just fto know , why fork freecad and not work for freecad directly ?

luzpaz, 2022-04-07:
ModernUI is a separate effort than Link branch, it was coded and then 'abandoned' by @hakanseven12. There is a forum thread dedicated to it where you can read more about it. 

realthunder created this branch as a proof of concept for several different endeavors. He codes at a breakneck speed and was limited by the slower pace of the upstream FreeCAD master. realthunder is not responsible to make every single upstream function of FreeCAD work in his branch. In fact that is counter productive. He does in fact contribute directly to upstream FreeCAD.  

Please remember when you open a ticket that you are going to utilize someone elses time and attention. Therefore provide them with as much details as you can. There is no deeper explanation here of what doesn't work specifically, what version of Link branch you're using (see [About](https://wiki.freecad.org/About), no screenshots, no error messages... please do better

luzpaz, 2022-04-07:
Closing this ticket. Please re-open if relevant info is presented

closed 2022-04-07 by luzpaz