realthunder/FreeCAD_assembly3 #1189: Illegal Storage access and crash

Neubo, 2022-04-06:
Hello,

In this post, I will speak about "**_new version_**" : 
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.406.26244 +4692 (Git)
Build type: Release
Branch: LinkDaily
Hash: 0c8e437ef38cdc5dd087529ce6825c6a2a130868
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)

and "**_old version_**" : 
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)

I would like to open a Step file in Freecad ([STI2k22.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8430138/STI2k22.zip))
With new version, freecad crash immediatly (Access Violation), right after open/import
With old version, it's ok and I can manipulate objects.

From the last version, I save the imported Step file to FCStd1 file.
When I open FCStd1 with new version, I can for eg. Rotate 20 seconds the object, then it crash also.
[image]

I read some post about this issue, but was corrected in old version. Looks like it's back again ! ;)
May you need more information/log, don't hesitate to ask, and give me the process to get them.

Thanks !

realthunder, 2022-04-07:
I can't reproduce the problem. Can you find `crash.dmp` file in your user directory (most likely `c:\Users\<username>\AppData\Roaming\FreeCAD`, and posted it here?

Neubo, 2022-04-07:
Sure, here it is.
[crash.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8441182/crash.zip)

I will try this evening to test the behaviour on another computer. Will let you know.

bambukouk, 2022-04-07:
Opens fine on my machine.
maybe Windows problem??

```
OS: elementary OS 6.1 JÃ³lnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.406.26244 +4692 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 0c8e437ef38cdc5dd087529ce6825c6a2a130868
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```
[image]

realthunder, 2022-04-08:
The `Cannot compute Inventor representation` error is due to another issue of not saving the shape properly with 01.11 release that is built with OCC 7.6. You can correct it by opening the old file in the latest release with OCC 7.6 and save again.

All Conda builds are using OCC 7.5. And the rest are using OCC 7.6 (Windows LibPack, MacOS Big Sur)

Neubo, 2022-04-08:
Thanks !

I've tested with today realease, everything works well.

Have a nice Week-end !

closed 2022-04-08 by Neubo