realthunder/FreeCAD_assembly3 #1190: Newest version crashing on many files 22.01.18 opened fine

Miksu80, 2022-04-07:
The 22.04.02 daily WIN64 crashes without any info on opening files. Find the failing parts here:
[Failing_parts.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8439116/Failing_parts.zip)

The stable version 22.04.02 file was corrupted 6.4.2022 and couldn't test these on that version..

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.402.26244 +4679 (Git)
Build type: Release
Branch: LinkDaily
Hash: 2ed36c0a234b3871a790ac8437f496f29e025ded
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/Finland (en_FI)

realthunder, 2022-04-07:
I can't reproduce the problem. Can you find `crash.dmp` file in your user directory (most likely `c:\Users\<username>\AppData\Roaming\FreeCAD`, and posted it here? Also, can you please show me your STEP import settings? Here is mine,

[image]

Miksu80, 2022-04-07:
Here you go:
[crash.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8439922/crash.zip)

Miksu80, 2022-04-08:
Works now with 2022.04.07

closed 2022-04-08 by Miksu80