realthunder/FreeCAD_assembly3 #1192: Problem with <<Create "{}">>

Kuzma30, 2022-04-07:
In translated version messages with <<Create "{}">> pattern show without last <<">>. Is it translating problem?

luzpaz, 2022-04-07:
Screenshot? About info?
Link to the translation string?

Kuzma30, 2022-04-07:
OS: Ubuntu Jammy Jellyfish (development branch) (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.30937 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.10.4
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.7.0.dev
Locale: Ukrainian/Ukraine (uk_UA)

[image]

luzpaz, 2022-04-07:
Can you link to the translation string in the source?

Kuzma30, 2022-04-07:
https://github.com/realthunder/FreeCAD_assembly3/blob/1d05e13540ac4cdedf3a382a6189bbf460150c3b/freecad/asm3/constraint.py#L917

closed 2022-04-08 by realthunder