realthunder/FreeCAD_assembly3 #1193: Solver using incorrect plane face for attachment and plane alignment

davesrocketshop, 2022-04-07:
The CAD problem I'm trying to solve is a nut fitting in an appropriate socket. In this case an M8 nut in a 3D printed part. I'm having some odd issues when attempting to apply constraints, particularly when it works on one side, close to the origin plane, but not on the other side away from the plane. The part on the opposite side is a mirrored part which may or may not be relevant.

First step is to align the nut with the hole using an axial alignment. Solve, and all is OK.

[image]

Next is to select two faces that will be fixed relative to each other, to prevent the nut from spinning as the part rotates.

[image]

If I try using a Plane Alignment, I get an error. I've tried moving the parts around so they're lined up, but same result.

[image]

Next I tried attaching the two faces with a surprising result. It flipped the nut and totally ignored the axial alignment constraint. No errors reported by the solver.

[image]

It appears it's trying to align with the reverse of the selected plane. In the case of the Plane Alignment, I suspect it's then detecting a conflict with the Axial constraint. In the case of the Attachment, it appears to be ignoring the axial constraint.

Sadly I didn't save before submitting the model and I  lost some work, but the attached model is an earlier version that exhibits the same behaviour.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27422 (Git)
Build type: Release
Branch: master
Hash: d938733eaf2c2ce7cb18d1cbb56147185c473530
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/Canada (en_CA)

[RudderAssembly.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8445029/RudderAssembly.zip)

davesrocketshop, 2022-04-07:
Also, I tried attaching the base of the nut to the bottom of the socket and it again inverted the nut embedding it inside the part.

jpg87Fr, 2022-04-07:
After constraining AxialAlignment, I would recommend Same Orientation rather than Plane Alignment to not overconstrain your assembly. As for the Attachment constraint, it is common to have to use the "Flip Element" or "Flip Part" contextual property.
See this page for more details :
http://help-freecad-jpg87.fr/020_ass_outils_contr.php#

davesrocketshop, 2022-04-07:
Thanks! Still learning this (obviously)

closed 2022-04-07 by davesrocketshop