realthunder/FreeCAD_assembly3 #1196: Can we start to make a distinction between A3 tickets and 'Link branch' tickets?

luzpaz, 2022-04-08:
The reason i ask is that there a several things going on:
* There are 2 repos: https://github.com/realthunder/FreeCAD/issues and https://github.com/realthunder/FreeCAD_assembly3
* A3 is now in the Addon Manager
* There are open tickets and PRs in https://github.com/realthunder/FreeCAD
* There are open tickets and PRs in  https://github.com/realthunder/FreeCAD_assembly3

I think a repo admin has the ability to move tickets between repos (if they have admin privs in each).

bambukouk, 2022-04-08:
Great idea :+1: 
Most of the people (especially on main forum) seem to be confused and equate `Assembly 3` with `Link branch` (wrong :-1: )

It is not helped by refusal of forum mods to create Link branch sub-forum and insisting on dumping anything remotely related to link branch into Assembly sub-forum.

Link branch is released in `FreeCAD_assembly3` and not in `FreeCAD link branch`

Most of us who have been around for a while manage OK, but to anybody coming fresh to it, it must be confusing...?

bambukouk, 2022-04-08:
It is indeed useful to make distinction, but...
humans will remain humans - trying to make things `fool proof ` is futile (it only leads to `better fools`)

as for ticket template it's a bit like "about FreeCAD" info requested to be included in every help question :-1: 
It has about 20% success rate
Regardless how big a font and how red it is, most of the people will ignore it ...

The only 100% solution is vetting and filtering by some moderator?
this is just my opinion (yes, I know everybody has an opinion on interweb..)

luzpaz, 2022-04-08:
@bambukouk it works a lot of the time in FreeCAD/FreeCAD so I'd say it's worth a shot. Train you userbase and they will eventually self-organize.

bambukouk, 2022-04-08:
Bottom line is - I totally agree with you about the need for distinction.
Let's get on with and we can sort out details later 👍

luzpaz, 2022-04-08:
No we just need to wait for the boss to weigh in :wink:

luzpaz, 2022-04-09:
@realthunder ping

realthunder, 2022-04-11:
I just tried transferring one of my own issue. It works find, but I didn't get any email notification. Looks like we need manually notified the author?

realthunder, 2022-04-11:
@luzpaz Do you have permission to move issue between this and FreeCAD Link branch?

luzpaz, 2022-04-11:
Will start doing this in the next few days :wink:

YosypC, 2022-04-11:
I've been wondering why there was a Linkstage3 AND an assembly3, AND why people often submit LS3 issues in the assembly3 Issue page, AND Linkstage3 releases are pushed in this GitHub instead of the LS3's.
I've followed the mass and began posting here too, but I guess it needs to change. Very confusing for newcomers.

luzpaz, 2022-04-11:
In progress

kevenwyld, 2022-04-15:
> Link branch is released in FreeCAD_assembly3 and not in FreeCAD link branch

In my opinion this is the root of most of the confusion.  Link branch installer/packages are released here in this repo and not realthunder/FreeCAD .  Is there a plan to change that?  Or is that even possible to change?

> A3 is now in the Addon Manager

While this is true, the link branch appimage seems to still be bundled with it.  So thinks are quite mixed together between these two repos and should probably be untangled further than just moving issues over.  In fact moving the issues away from the releases might make things even more confusing in the interim.

Also is the A3 addon the same as the one bundled with the link branch or is it lagging behind because it is built separately?

luzpaz, 2022-04-15:
Indeed. We're slowly teasing apart the issue. There are so many posts/link/social-media mentions that point to A3 release page for Link branch builds. But it's realthunder's call in the end.

ceremcem, 2022-06-24:
@luzpaz I didn't wery well understand the desired workflow. Are you suggesting we should open Assembly3_WB related issues here and LinkStage3 (FreeCAD itself) issues in the `realthunder/FreeCAD` repo?

If that's the case, I see a slight problem here. A single bug might be related both with LinkStage3 and Assembly3_WB or even with another WB (like SheetMetal, remember realthunder's commit to SheetMetal for that WB to work with LinkStage3); it would be hard to determine where to open the issue.

I think we should have opened every issue in Assembly3_WB issue tracker, no matter what it is related to.

luzpaz, 2022-06-24:
The point is to start disentangling A3 from Link branch since A3 can run independently from Addon Manager in master. So yes, it's a bit messy but it's a start

bob-u, 2022-11-13:
What's confusing to me is this:

Do Link branch releases have all Assembly 3 patches?