realthunder/FreeCAD_assembly3 #1197: Chamfer

hamza00789, 2022-04-09:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.406.26244 +4692 (Git)
Build type: Release
Branch: LinkDaily
Hash: 0c8e437ef38cdc5dd087529ce6825c6a2a130868
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)
 
The chamfer is not working on this edge. The result is very bad.  

[image] 

[image] 

[image]  

You can see the problem in the file. 

[Chamfer.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8456567/Chamfer.zip)

herrgerd, 2022-04-09:
Maybe use the traditional way with the Groove-Tool instead of sectioning/slicing, hiding one part and...
[image]

bambukouk, 2022-04-09:
Simple - the issue is coincident edges
Rotate/Offset Sketch001 attachment around Y axis away from the edge and all is well:
[image]

hamza00789, 2022-04-09:
Can you post this file? 

[image]

closed 2022-04-09 by hamza00789