realthunder/FreeCAD_assembly3 #1198: SIGSEGV when try to fix broken link.

Kuzma30, 2022-04-09:
First, the original assembly (screw + nut) was in the "main" file. Then it was decided to move all fasteners into one file. Therefore, a separate file with the assembly was created. In a "large" assembly, the screw+nut assembly was removed, leaving an LinkArray (with question marks). When I tried to drag the assembly from a separate file (to update the link) to main assembly, I got the following error.

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fc7c668a520]
#1  0x7fc7c8b000f8 in App::PropertyLinkBase::tryReplaceLink[abi:cxx11](App::PropertyContainer const*, App::DocumentObject*, App::DocumentObject const*, App::DocumentObject*, App::DocumentObject*, char const*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADApp.so+0x178
#2  0x7fc7c8b13234 in App::PropertyLinkBase::tryReplaceLinkSubs(App::PropertyContainer const*, App::DocumentObject*, App::DocumentObject const*, App::DocumentObject*, App::DocumentObject*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADApp.so+0x1a4
#3  0x7fc7c8b1356c in App::PropertyXLink::CopyOnLinkReplace(App::DocumentObject const*, App::DocumentObject*, App::DocumentObject*) const from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADApp.so+0x4c
#4  0x7fc7c99194d3 in Gui::ViewProviderDocumentObject::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x393
#5  0x7fc7c997d0d4 in Gui::ViewProviderLink::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x3b4
#6  0x7fc7c998c543 in Gui::ViewProviderPythonFeatureT<Gui::ViewProviderLink>::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x63
#7  0x7fc7c992b5d3 in Gui::ViewProviderPy::replaceObject(_object*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x93
#8  0x7fc7c992b679 in Gui::ViewProviderPy::staticCallback_replaceObject(_object*, _object*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x59
#9  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12b048) [0x7fc7c6be7048]
#10  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7fc7c6ba1d9c]
#11  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dcc) [0x7fc7c6b37a0c]
#12  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c61ff) [0x7fc7c6c821ff]
#13  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0xe7188) [0x7fc7c6ba3188]
#14  0x7fc7c9942a78 in Gui::ViewProviderPythonFeatureImp::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x1a8
#15  0x7fc7c998c52b in Gui::ViewProviderPythonFeatureT<Gui::ViewProviderLink>::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x4b
#16  0x7fc7c96cefeb in Gui::TreeWidget::dropEvent(QDropEvent*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x531b
#17  0x7fc7c96c8ae5 in Gui::TreeWidget::eventFilter(QObject*, QEvent*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x685
#18  0x7fc7c75b2a9e in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x8e
#19  0x7fc7c80a2750 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xc0
#20  0x7fc7c941a400 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x90
#21  0x7fc7c75b2e7a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#22  0x7fc7c7999307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#23  0x7fc7c796ea2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#24  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7fc7bc35bd6e]
#25  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7fc7c49d5d1b]
#26  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xaa6f8) [0x7fc7c4a2a6f8]
#27  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fc7c49d33c3]
#28  0x7fc7c760c0f8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#29  0x7fc7c75b179b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#30  0x7fc7c75b9d34 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#31  0x7fc7c93649fe in Gui::Application::runApplication() from /home/aleksandr/FreeCAD-build-master/lib/libFreeCADGui.so+0x41e
#32  ./FreeCAD-build-master/bin/FreeCAD(+0x50d4) [0x564cf748c0d4]
#33  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fc7c6671d90]
#34  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fc7c6671e40]
#35  ./FreeCAD-build-master/bin/FreeCAD(+0x54e5) [0x564cf748c4e5]
`

bambukouk, 2022-04-09:
just out of curiosity - is FreeCAD you are using your own version?
I am confused by some elements (highlighted below):
Version: `0.20.30938 (Git)`
Build type: Release
Branch: LinkDaily
OCC version: `7.7.0.dev`

Kuzma30, 2022-04-09:
Yes, I compile it with dev branch of OCCT.

Kuzma30, 2022-04-10:
`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f74e3daa520]
#1  0x7f74e62200f8 in App::PropertyLinkBase::tryReplaceLink[abi:cxx11](App::PropertyContainer const*, App::DocumentObject*, App::DocumentObject const*, App::DocumentObject*, App::DocumentObject*, char const*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADApp.so+0x178
#2  0x7f74e6233234 in App::PropertyLinkBase::tryReplaceLinkSubs(App::PropertyContainer const*, App::DocumentObject*, App::DocumentObject const*, App::DocumentObject*, App::DocumentObject*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADApp.so+0x1a4
#3  0x7f74e623356c in App::PropertyXLink::CopyOnLinkReplace(App::DocumentObject const*, App::DocumentObject*, App::DocumentObject*) const from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADApp.so+0x4c
#4  0x7f74e70394b3 in Gui::ViewProviderDocumentObject::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x393
#5  0x7f74e709d0b4 in Gui::ViewProviderLink::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x3b4
#6  0x7f74e70ac523 in Gui::ViewProviderPythonFeatureT<Gui::ViewProviderLink>::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x63
#7  0x7f74e704b5b3 in Gui::ViewProviderPy::replaceObject(_object*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x93
#8  0x7f74e704b659 in Gui::ViewProviderPy::staticCallback_replaceObject(_object*, _object*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x59
#9  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12b048) [0x7f74e4307048]
#10  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7f74e42c1d9c]
#11  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dcc) [0x7f74e4257a0c]
#12  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c61ff) [0x7f74e43a21ff]
#13  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0xe7188) [0x7f74e42c3188]
#14  0x7f74e7062a58 in Gui::ViewProviderPythonFeatureImp::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x1a8
#15  0x7f74e70ac50b in Gui::ViewProviderPythonFeatureT<Gui::ViewProviderLink>::replaceObject(App::DocumentObject*, App::DocumentObject*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x4b
#16  0x7f74e6deefcb in Gui::TreeWidget::dropEvent(QDropEvent*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x531b
#17  0x7f74e6de8ac5 in Gui::TreeWidget::eventFilter(QObject*, QEvent*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x685
#18  0x7f74e4cd2a9e in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x8e
#19  0x7f74e57c2750 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xc0
#20  0x7f74e6b3a3e0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x90
#21  0x7f74e4cd2e7a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#22  0x7f74e50b9307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#23  0x7f74e508ea2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#24  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7f74d9a98d6e]
#25  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7f74e20f6d1b]
#26  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xaa6f8) [0x7f74e214b6f8]
#27  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f74e20f43c3]
#28  0x7f74e4d2c0f8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#29  0x7f74e4cd179b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#30  0x7f74e4cd9d34 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#31  0x7f74e6a849de in Gui::Application::runApplication() from /home/aleksandr/FreeCAD-build-V7_6_1/lib/libFreeCADGui.so+0x41e
#32  ./FreeCAD-build-V7_6_1/bin/FreeCAD(+0x50d4) [0x5585451ef0d4]
#33  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f74e3d91d90]
#34  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f74e3d91e40]
#35  ./FreeCAD-build-V7_6_1/bin/FreeCAD(+0x54e5) [0x5585451ef4e5]
`
OS: Ubuntu Jammy Jellyfish (development branch) (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.30938 (Git)
Build type: Release
Branch: LinkDaily
Hash: 49c51c4b88c5f876b1aca4fa22c96281e8b9ed86
Python version: 3.10.4
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.6.1
Locale: Ukrainian/Ukraine (uk_UA)

luzpaz, 2022-04-10:
Note: you are using Python 3.10.x and there are some issues I've seen reported w/ 3rd party deps (perhaps coin or pivy)?

Kuzma30, 2022-04-10:
> Note: you are using Python 3.10.x and there are some issues I've seen reported w/ 3rd party deps (perhaps coin or pivy)?

I will try with older Python later.

Kuzma30, 2022-04-10:
Try on Windows machine

OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Ukrainian/Ukraine (uk_UA)
  
Same try to drag and drop from external file "screw + nut" assembly.
```
17:36:59  Traceback (most recent call last):
  File "C:\Users\kuzem\AppData\Roaming\FreeCAD\Mod\FreeCAD_assembly3\freecad\asm3\assembly.py", line 525, in replaceObject
    self.ViewObject.replaceObject(oldObj,newObj)
<class 'Base.FreeCADError'>: Access violation
17:36:59  <Exception> Tree.cpp(4963): Failed to replace object Link029 with Assembly005
```

On 
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Ukrainian/Ukraine (uk_UA)

Error looks like 
```17:43:13  Illegal storage access...
17:43:13  Traceback (most recent call last):
  File "C:\Users\kuzem\AppData\Roaming\FreeCAD\Mod\FreeCAD_assembly3\freecad\asm3\assembly.py", line 525, in replaceObject
    self.ViewObject.replaceObject(oldObj,newObj)
<class 'Base.FreeCADError'>: Illegal storage access! Please save your work under a new file name and restart the application!
17:43:13  <Exception> Tree.cpp(4963): Failed to replace object Link029 with Assembly005```

realthunder, 2022-04-11:
I can't reproduce the problem. Can you please provide the files just before dragging that caused the crash?

Kuzma30, 2022-04-11:
Files here
https://github.com/Kuzma30/FREECAD_CNC
Main assembly is "СтанинаСборка" 
[image]
Then open "Крепеж"
[image]

closed 2022-04-11 by realthunder