realthunder/FreeCAD_assembly3 #1200: Link arrays in assemblies make move tools inaccessible

shyblower, 2022-04-11:
When I use link arrays inside of an assembly I cannot utilize any of the move tools (move part, axial move part, quick move) for any part in the same assembly. They become greyed out and inaccessible.

realthunder, 2022-04-11:
I can't reproduce the problem. Can you please post an demo file showing the problem. A screen cast will be better, too.

shyblower, 2022-04-12:
It happens when you constrain an array element, regardless of the constraint type. After that the three tools become greyed out and inaccessible for every part in the assembly. As long as you just constrain non array parts everything works as usual.
If you still need a screen cast I can provide one, but I believe you will be able to reproduce it now.

realthunder, 2022-04-12:
Please post your FreeCAD about information here. And yes please post the screen cast. Maybe I misunderstood what you're saying, but I still can't reproduce the problem.

https://user-images.githubusercontent.com/1207888/162896781-4c842f0e-0be7-4787-9456-fd7a67fd3aa7.mp4

realthunder, 2022-04-12:
Did you manually install Assembly3 workbench before by cloning this repo to your user FreeCAD directory? If so, please delete the directory and restart FreeCAD. I suspect you're using an older version.

realthunder, 2022-04-12:
Oh, now that I rewatched your video, I can see where the problem is. You have the `Lock mover` option enabled. It seems there is a bug in that option for Link array. I'll fix it soon. You can disable that option for a work around.

shyblower, 2022-04-12:
Indeed, it was the lock mover. Thank you for investigating this, and for your efforts regarding FreeCAD development.

realthunder, 2022-04-12:
You're welcome. It's been fixed by the above commit.