realthunder/FreeCAD_assembly3 #1201: Unable to add constraints when using Assembly 3 workbench

troetter, 2022-04-11:
I've been trying to create a simple assembly of a few 3d printed parts and some off-the-shelf bearings and fasteners. I've encounter numerous frustrating issues that I don't understand the reasons behind. I'm using the `freecad-realthunder` snap with version string "2022-02-13-gce7fe6dc" on Fedora 35.

I have attached a zip with three files. In 'slide_assy' I'm trying to assemble bodies from the other two files. The current issue is that I want to add an instance of a washer and constrain it with an axial and a planar alignment.

Step by step, reproducible on my end:

1. Open 'slide_assy' document
2. Switch to Assembly 3 wb
3. Open 'cots' document
4. With 'Assembly' under 'slide_assy' enabled, drag 'Wiberger-din125_8-4' into 'Assembly'
5. Select left-most hole surface in 'SlidePart1'
6. Ctrl select inner surface of washer
7. Press 'Create "AxialAlignment" constraint'

The error reported is:
```
18:28:12  173.603949 <asm3.sys> system.py(244): auto relax constraint (slide_assy#AxialAlignment001) slide_assy#Constraint002 between (slide_assy#SlidePart2) slide_assy#Link001 and (slide_assy#SlidePart1) slide_assy#Link, 2
18:28:12  373.365058 <asm3.main> assembly.py(3918): solver exception when auto recompute
Traceback (most recent call last):
  File "/snap/freecad-realthunder/8/usr/Mod/Assembly3/freecad/asm3/solver.py", line 110, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "/snap/freecad-realthunder/8/usr/Mod/Assembly3/freecad/asm3/sys_slvs.py", line 70, in solve
    raise RuntimeError(reason)
RuntimeError: inconsistent constraints

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 535, in _catch
  File "/snap/freecad-realthunder/8/usr/Mod/Assembly3/freecad/asm3/solver.py", line 412, in solve
    return _solve(*args,**kargs)
  File "/snap/freecad-realthunder/8/usr/Mod/Assembly3/freecad/asm3/solver.py", line 382, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/snap/freecad-realthunder/8/usr/Mod/Assembly3/freecad/asm3/solver.py", line 113, in __init__
    raise RuntimeError(translate('asm3', 'Failed to solve {}: {}').format(
RuntimeError: Failed to solve slide_assy#Assembly: inconsistent constraints
```

While trying to create this assembly I have encountered numerous similar issues, where it is impossible to add a constraint. Sometimes it has worked if I first move the newly linked part before trying to add the constraint. Other times it has worked if I link the part, save and close all files and restart before trying to add any constraints. I haven't found a pattern however, and I haven't found anything that works in this particular case.

[throttle_testbed.zip](https://github.com/realthunder/FreeCAD/files/8147097/throttle_testbed.zip)

troetter, 2022-05-13:
I did not notice until now that this issue had been transferred here. I see also that "due to overconstraining" has been added to the title. I still don't understand how I can be overconstraining when I'm trying to add the first constraint to the newly added part.

luzpaz, 2022-07-01:
@troetter please open a thread on the [Freecad forum](https://forum.freecad.org/viewforum.php?f=20) asking for orientation around this.