realthunder/FreeCAD_assembly3 #1203: installing error

hamza00789, 2022-04-13:
This error occurred while installing sheet metal add-ons. Is this a problem? 

11:51:39  GitPython not found. Using standard download instead.
11:51:40  Traceback (most recent call last):
11:51:40    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\Mod\AddonManager\addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
11:51:40    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
11:51:40    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 531, in open
    response = meth(req, response)
11:51:40    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 640, in http_response
    response = self.parent.error(
11:51:40    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 569, in error
    return self._call_chain(*args)
11:51:40    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
11:51:40    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
11:51:40  urllib.error.HTTPError: HTTP Error 404: Not Found
11:51:40  Downloading https://github.com/shaise/FreeCAD_SheetMetal/archive/master.zip

hamza00789, 2022-04-13:
Now this error message comes when installing any workbench   

12:00:37  GitPython not found. Using standard download instead.
12:00:38  Traceback (most recent call last):
12:00:38    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\Mod\AddonManager\addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
12:00:38    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
12:00:38    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 531, in open
    response = meth(req, response)
12:00:38    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 640, in http_response
    response = self.parent.error(
12:00:38    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 569, in error
    return self._call_chain(*args)
12:00:38    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
12:00:38    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
12:00:38  urllib.error.HTTPError: HTTP Error 404: Not Found
12:00:38  Downloading https://github.com/mnesarco/FreeCAD_ExtMan/archive/master.zip
12:01:14  GitPython not found. Using standard download instead.
12:01:15  Traceback (most recent call last):
12:01:15    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\Mod\AddonManager\addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
12:01:15    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
12:01:15    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 531, in open
    response = meth(req, response)
12:01:15    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 640, in http_response
    response = self.parent.error(
12:01:15    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 569, in error
    return self._call_chain(*args)
12:01:15    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
12:01:15    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
12:01:15  urllib.error.HTTPError: HTTP Error 404: Not Found
12:01:15  Downloading https://github.com/shaise/FreeCAD_SheetMetal/archive/master.zip
12:01:36  GitPython not found. Using standard download instead.
12:01:37  Traceback (most recent call last):
12:01:37    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\Mod\AddonManager\addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
12:01:37    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
12:01:37    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 531, in open
    response = meth(req, response)
12:01:37    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 640, in http_response
    response = self.parent.error(
12:01:37    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 569, in error
    return self._call_chain(*args)
12:01:37    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
12:01:37    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
12:01:37  urllib.error.HTTPError: HTTP Error 404: Not Found
12:01:37  Downloading https://github.com/easyw/Manipulator/archive/master.zip

bambukouk, 2022-04-13:
This is not `assembly` issue.
Belongs here: https://github.com/realthunder/FreeCAD/issues
as per https://github.com/realthunder/FreeCAD_assembly3/issues/1196#issue-1197600774

realthunder, 2022-04-13:
So did the workbench installed properly?

Please try start freecad with the `RunFreeCAD.bat` file in your archive extracted directory and try again.

hamza00789, 2022-04-13:
> So did the workbench installed properly?
 
Yes, the workbench was installed properly but I wanted to point out this error.

hamza00789, 2022-04-13:
> Please try start freecad with the `RunFreeCAD.bat` file in your archive extracted directory and try again. 

I started freecad with Run Freecad.bat but the error message is still coming. 

Traceback (most recent call last):
12:37:54    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\Mod\AddonManager\addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
12:37:54    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
12:37:54    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 531, in open
    response = meth(req, response)
12:37:54    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 640, in http_response
    response = self.parent.error(
12:37:54    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 569, in error
    return self._call_chain(*args)
12:37:54    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
12:37:54    File "C:\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407\bin\lib\urllib\request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
12:37:54  urllib.error.HTTPError: HTTP Error 404: Not Found
12:37:54

hamza00789, 2022-04-13:
This error message comes up every time I use a part container. 

13:10:46  <MDIView> ActiveObjectList.cpp(216): Cannot set active object Unnamed#Part. in document 'Unnamed'. Not found in current selection
13:10:46  Traceback (most recent call last):
13:10:46    File "C:\Users\hamza\AppData\Roaming\FreeCAD\Mod\Part-o-magic\PartOMagic\Gui\Observer.py", line 242, in poll
13:10:46      self.trackActiveContainer()
13:10:46    File "C:\Users\hamza\AppData\Roaming\FreeCAD\Mod\Part-o-magic\PartOMagic\Gui\Observer.py", line 283, in trackActiveContainer
13:10:46      assert(ac is new_ac)
13:10:46  AssertionError 
 
[mp4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8479439/mp4.zip)

hamza00789, 2022-04-13:
> This error message comes up every time I use a part container.   

The part-container error message disappears when uninstalling Part-o-Magic.

hamza00789, 2022-04-13:
Can you tell me how to get rid of these error messages?

luzpaz, 2022-04-13:
This ticket is disorienting. There is no [about](https://wiki.freecad.org/About) info which would tell us what addons are installed as well. 

Checkout:  https://github.com/DeepSOIC/Part-o-magic#uninstall

hamza00789, 2022-04-13:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

hamza00789, 2022-04-13:
[image] 

I have these addons installed. Would you like to ask me more about this?

hamza00789, 2022-04-13:
There are two videos, one with part o magic and one without part o magic. As can be seen in the second video, Part o Magic is not installed. And it can be seen that there was no error message while taking part container. 
This makes it clear. The error message that occurs when taking part containers is due to part o magic. 

[1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8482298/1.zip) 

[2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8482301/2.zip)

luzpaz, 2022-04-13:
> I have these addons installed. Would you like to ask me more about this?

In upstream master, About info also shows installed addons/macros. Thought that Link branch had that but I was wrong.

luzpaz, 2022-04-13:
> [1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8482298/1.zip)
> 
> [2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8482301/2.zip)

can you not add the movies directly instead of zipping?

hamza00789, 2022-04-13:
In that case, you will not be able to watch the video. I don't know why but I have this problem now. You can watch one of my videos by posting it without zipping.

bambukouk, 2022-04-13:
`Note: Video codec compatibility is browser specific, and it's possible that a video you upload to one browser is not viewable on another browser. At the moment we recommend using h.264 for greatest compatibility.`

as per:
https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/attaching-files

and yes, your videos attached directly don't show in preview

luzpaz, 2022-04-13:
@DeepSOIC would you be so kind to weigh-in on this ? (since it could be part-o-magic related)

DeepSOIC, 2022-04-13:
I highly doubt the original issue has anything to do with part-o-magic. The errors when new Part is created certainly do. I don't immediately know the cause, i need to check it out, but i'm not sure if i'm gonna do it any time soon... I've never tested PoM for compatibility with realthunder's branch.

realthunder, 2022-04-14:
The problem is fixed by the above commit. This is harmless error. But you want to fix it now, just find `addonmanager_utilities.py` and `addonmanager_workers.py` and manually change it as shown in the commit.

closed 2022-04-14 by realthunder