realthunder/FreeCAD_assembly3 #1208: question _axial alignment

bambukouk, 2022-04-15:
Simple assembly.
Box with a round hole and round peg.
Two cylindrical faces used as elements.
[image]

When trying to constrain using Axial alignment, get error message (below).
What am I doing wrong, please?

```
16:30:23  514.915552 <asm3.main> assembly.py(3919): solver exception when auto recompute
Traceback (most recent call last):
  File "/tmp/.mount_FreeCA2SHC2J/usr/Mod/asm3/freecad/asm3/solver.py", line 110, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "/tmp/.mount_FreeCA2SHC2J/usr/Mod/asm3/freecad/asm3/sys_slvs.py", line 70, in solve
    raise RuntimeError(reason)
RuntimeError: inconsistent constraints

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 535, in _catch
  File "/tmp/.mount_FreeCA2SHC2J/usr/Mod/asm3/freecad/asm3/solver.py", line 412, in solve
    return _solve(*args,**kargs)
  File "/tmp/.mount_FreeCA2SHC2J/usr/Mod/asm3/freecad/asm3/solver.py", line 382, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/tmp/.mount_FreeCA2SHC2J/usr/Mod/asm3/freecad/asm3/solver.py", line 113, in __init__
    raise RuntimeError(translate('asm3', 'Failed to solve {}: {}').format(
RuntimeError: Failed to solve (axial_alignment_test#Assembly) Unnamed#Assembly: inconsistent constraints 
```

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

File:
[axial_alignment_test.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8496772/axial_alignment_test.FCStd.zip)

realthunder, 2022-04-16:
I can reproduce the problem. It may fail or not depending on whether to select the box face first or the cylinder face. And when it is not moved correctly, there may either be warning message about redundant constraint or error message about inconsistent constraint. I think the problem is caused by the initial position of the parts being too far apart, and the solver is confused about what message to output. As a general 'solution' to this kind  of problem, just manually move the parts a bit closer.

bambukouk, 2022-04-16:
Thank you for the confirmation.
That was my understanding as well, but I was beginning to doubt my sanity after questions on the subject on main forum.

closed 2022-04-16 by bambukouk