realthunder/FreeCAD_assembly3 #1212: The offset command is not working properly.

hamza00789, 2022-04-16:
The offset command is not working properly. The result is very bad. You can see how the surface is cracking. There is nothing wrong with the surface that is being offset (geometry results fine). 

https://user-images.githubusercontent.com/92737149/163663208-34ef7166-d7ae-41f8-a9b4-8ca923525fb0.mp4

bambukouk, 2022-04-16:
https://forum.freecadweb.org/viewtopic.php?f=8&t=68040

So, it is not working in both Link branch and Master branch?
BTW why is it here in Assembly 3 issues?

It is Part workbench problem?

hamza00789, 2022-04-16:
>BTW why is it here in Assembly 3 issues?
> 
> It is Part workbench problem? 

I don't know where the problem should be.Can you guide me on where the issue of the part workbench should be?

bambukouk, 2022-04-16:
Offset is a tool in Part workbench (that's what I meant, rather than suggesting that the problem is in the actual workbench)

I understand that it is OCC related and therefore not possible to address here?

luzpaz, 2022-04-16:
@hamza00789 just FYI https://github.com/realthunder/FreeCAD_assembly3/issues/1196